KlipperScreen/KlipperScreen #583: zcalibrate / manual delta calibrate method=manual  without "probe defined" enclosed is a video. request

danorder, 2022-04-10:
https://user-images.githubusercontent.com/75015184/162603659-6b2b5c7f-acf0-4eed-9bdc-89bb10b8b189.mp4

danorder, 2022-04-10:
follow up to this. the flsun printers require this in some cases. another reason is simply if a probe breaks or, is  lost  a simple  fall back remains a option. Alot of people using these printers are "new" to delta or klipper the buttons keep it simple / similar to stock functionality. in other cases the probes are not accurate enough or lined up quite right to the  nozzle. manual alleviates this. lastly its just a pain if the pc is not near by to run a feeler guage.  with the tft this can be done in a few minutes.

alfrix, 2022-04-10:
but's it's already there 4bace34847ee696178671f7af2da4b00f71e0900
you probably need to update
can you check your ks version in the logfile? or provide the logfile so i can check it

danorder, 2022-04-10:
i have it all the panel says check the log. which i already submitted and,
nothing turned up. the panel is completly disabled. i seen a mention of
probe required for confing.  but delta calibrate manual doesn't use a
[probe] definition. i can send another one. thus far in discord ive done
the wipe klipper screen , reset git repo to no avail. reverting 10 days
back , everything works as "normal"  feedrate , and extrusion panels don't
output anything that i can see either.

alfrix, 2022-04-10:
>i already submitted

no you didn't, this issue is another issue, and the other log didn't show anything

danorder, 2022-04-11:
the other log is going to be the same. i can reinstall it again, and see if
it spits out something different. the tft buttons don't even show the
percent changing. what is the current recommended iso to use? it may be
worth just wiping out the whole thing since none of the listed
troubleshooting seems to have any effect.

danorder, 2022-04-11:
I also forgot to mention. yes moonraker / klipper etc are all latest.

danorder, 2022-04-12:
2022-04-12 00:42:36,356 [screen.py:_load_panel()] - Loading panel: zcalibrate
2022-04-12 00:42:36,356 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/zcalibrate.py
2022-04-12 00:42:36,397 [screen.py:show_panel()] - Unable to load panel zcalibrate
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 304, in show_panel
    self.panels[panel_name].initialize(panel_name)
  File "/home/pi/KlipperScreen/panels/zcalibrate.py", line 47, in initialize
    if not self._screen.printer.get_config_section("stepper_z")['endstop_pin'].startswith("probe"):
TypeError: 'bool' object is not subscriptable
2022-04-12 00:42:36,400 [screen.py:show_error_modal()] - Showing error modal: Unable to load panel zcalibrate
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 304, in show_panel
    self.panels[panel_name].initialize(panel_name)
  File "/home/pi/KlipperScreen/panels/zcalibrate.py", line 47, in initialize
    if not self._screen.printer.get_config_section("stepper_z")['endstop_pin'].startswith("probe"):
TypeError: 'bool' object is not subscriptable

closed 2022-04-12 by alfrix

danorder, 2022-10-11:
i should also mention it keeps wanting a newer pip? is that now required or
to be ignored