KlipperScreen/KlipperScreen #584: Print Status: Temp of Chamber

4noxx, 2022-04-10:
hello

how can i add the temperature of my chamber on this screen?

[image]

alfrix, 2022-04-10:
current;y you can't have it there, i'm working on some updates, but it will take a while

for now, use the titlebar https://klipperscreen.readthedocs.io/en/latest/Configuration/#printer-options

4noxx, 2022-04-10:
like this?

[image]

4noxx, 2022-04-10:
hmm, i see only the printername on the top of the screen

[image]

[image]

4noxx, 2022-04-10:
works now. 
think IP and Port are necessary
```
[printer Anet AM8]

moonraker_host: 127.0.0.1
moonraker_port: 7125
moonraker_api_key: False

titlebar_items: chamber, MCU, Pi

titlebar_name_type: None

use_dpms: True

```

alfrix, 2022-04-11:
no they aren't,  the titlebar temp items don't load while printing, it's a bug

also use_dpms doesn't go under [printer] it goes under [main] do not put things blindly in your config please read and use what you need

PR #592 (2022-05-01): Improve Job status
Adds heater_generic and/or temp_sensors besides extruder/bed (because of the limited space maybe only 1)
Temps are now buttons and act as a shortcut to the temp panel
Adds fan_generic to the fan label and it's now a shortcut to the fan panel
The colors and size of the progress circle were changed to be more subtle.
Margins between items were augmented, because they are buttons now
Adds 3 information pages: move, extrusion and time:
* move can be opened with speed or z buttons
* time with elapsed/remaining buttons
* extrusion with the extrude_factor / flowrate button

TODO
-  [... cut]
closed 2022-05-01 by alfrix