KlipperScreen/KlipperScreen #593: xserver-xorg-video-fbturbo dependency not installable

th33xitus, 2022-04-21:
I came across an issue with `xserver-xorg-video-fbturbo` not available on Debian 11, Ubtunu 21.10 Server and Ubuntu 20.04 Server.
Is this a package only available in the raspi package list?

The KlipperScreen installer doesn't offer alternative packages for other distros and the installation of dependencies will **silenty fail**, still the script will continue to run. It then fails when it comes to creating the virtualenv and building wheels for pycairo and some other python packages. Building fails, because the dependencies for building them won't get installed due to the earlier mentioned silent fail of https://github.com/jordanruthe/KlipperScreen/blob/master/scripts/KlipperScreen-install.sh#L32
For example, afaik for building pycairo `libcairo2-dev` is required. This won't happen -> building the wheels fails.

It would be nice to work out a solution for that. I am the dev behind KIAUH and you contributed to my project with implementing KlipperScreen so people can easily install it with KIAUHs help.

Since im not sure which Linux distro KlipperScreen targets (do you target only Raspberry Pi OS ? - i searched your docs but couldn't find any limitations mentioned) im also fine with working out a solution myself if you are not interested in establishing compatibility of your own KlipperScreen install script with other distros. In that case it would be nice to get to know **which** other package can be installed as a replacement for `xserver-xorg-video-fbturbo` on Debian, Ubuntu etc.

alfrix, 2022-04-22:
> Is this a package only available in the raspi package list?

No, but it may not exist on other distros or just may have a different name like xf86-video-fbturbo

> ...will silenty fail.

i will prepare an update to the installer with extra safeguards.

> do you target only Raspberry Pi OS

yes, but users have shown debian/ubuntu working, i will not be testing a bunch of different distros.

> which other package can be installed as a replacement for xserver-xorg-video-fbturbo

being the video driver it depends on the hardware, it may be xserver-xorg-video-fbdev or other.

th33xitus, 2022-04-22:
> yes, but users have shown debian/ubuntu working, i will not be testing a bunch of different distros.

fair enough.

i will wait for the changes you make :)

PR #601 (2022-04-29): installer improvements
The installer now catchs apt errors
FBturbo is now optional and FBdev is used as fallback
install xserver-xorg-legacy but do not force it, it's usually required for a lot of screens, but not 100%
hide pip update warnings, the new version broke stuff.
more granularity and steps to what's being installed, should make it easier to debug if something goes wrong

fixes #593 
closes #604
closed 2022-04-29 by alfrix