KlipperScreen/KlipperScreen #594: Zbolt theme - it is hard to see the selections on the fine tuning pages

simpat1zq, 2022-04-22:
I have 2 different types of screens. One is the official pieces display and the other 2 are hdmi/usb. When I'm on the fine tuning page while using the zbolt theme, it is very difficult to see the which adjustment is selected at the bottom. On the Pi display, I can only tell if I'm looking directly in front if the display. If I'm at a slight angle, the color looks the same as the unselected values. 

On the hdmi displays, I can barely tell the difference, no matter the angle. I think it might be better to use a different color for the selected value. Or is there a way to make the change myself? I couldn't find the docs on the themes.

alfrix, 2022-04-23:
your hardware has bad viewing angles, use another theme or copy it with another name and modify it, i will not change it, since the theme is maintained by Jordan and i leave the the decision to him

beroliv, 2023-08-10:
i have the same issue with all the themes, very difficult to see wich button is active.
i

commit f5a5efab (2023-10-21): styles: use gtk vars for colors and other cleanup and improvements close #594
closed 2023-10-21 by alfrix