KlipperScreen/KlipperScreen #597: DSI Touch not working

burnsy180, 2022-04-22:
I bought a 4.3" WaveShare DSI LCD screen for my CR10S Pro V2 as I can't use the stock screen with the latest Klipper version.

I've installed MainsailOS, Klipper and Klipperscreen on my Pi4B per the docs, and Klipperscreen shows up great on the screen. However the touch function does not work.
I can connect a USB mouse to the Pi and click through menus fine (blindly without a cursor), but no touchscreen.

Given that this screen is listed as 'known working', and everyone else has said theirs was just plug-and-play, is there some sort of configuration I can do to fix/enable the touch function?

I've gone through the troubleshooting on both Klipperscreen docs and the WaveShare site (enabling the L2 driver), but they make no difference.

Logs attached
[Xorg.0.log](https://github.com/jordanruthe/KlipperScreen/files/8540649/Xorg.0.log)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/8540653/KlipperScreen.log)
.

yzheka, 2022-04-22:
I got the same issue as yours. I think it is not KlipperScreen issue because when I installed older version of fluiddpi image it started to work. and it is working after I made full update all to the latest versions.

alfrix, 2022-04-23:
this is not a ks problem, when in doubt test your hardware with other operating system like desktop raspbian, it's probably a bad connection with the cable

closed 2022-04-23 by alfrix