KlipperScreen/KlipperScreen #598: Feature request: message on screen on shutdown

sprior, 2022-04-22:
It would be nice and helpful if after doing an OS shutdown something could be put in place to clear the screen and put up a full screen message that it is safe to power off.  I haven't gotten too far into the details yet but this might be a clue:

https://unix.stackexchange.com/questions/39226/how-to-run-a-script-with-systemd-right-before-shutdown

alfrix, 2022-04-23:
i don't think this can be done, the driver is unloaded meaning you won't see anything on screen and the device would be continuing with the shutdown, just wait until the green led stops blinking

closed 2022-04-23 by alfrix

sprior, 2022-04-23:
I didn't mean a graphical screen as I know the driver is unloaded.  I meant a text screen - clearing the screen and printing SHUT DOWN in console text would be sufficient, bonus points if it's centered in the middle of the screen.