KlipperScreen/KlipperScreen #599: The fan screen is unusable and not visible.

AntoszHUN, 2022-04-23:
I attached the pictures that do not show the fan control screen. This menu is simply useless.
LOG file not found in tmp folder.
[image]
[image]

alfrix, 2022-04-23:
i cannot fix it without a logfile

AntoszHUN, 2022-04-23:
> 

> i cannot fix it without a logfile

KlipperScreen does not create a log file. Maybe there's an order for it? I can't find the KlipperScreen.log file anywhere.

AntoszHUN, 2022-04-23:
if I think Klippy logos KlipperScreen? Here is the log file.
[klippy.log](https://github.com/jordanruthe/KlipperScreen/files/8546978/klippy.log)

alfrix, 2022-04-23:
it's not the same, but it helped, this is caused by the name of the fan, you need to change the name of the fan to something without those special characters or it will not update

fixed crash with a77134d63b4895bfbc6a5e460b166dae261fed73

closed 2022-04-23 by alfrix

AntoszHUN, 2022-04-23:
> 

> it's not the same, but it helped, this is caused by the name of the fan, you need to change the name of the fan to something without those special characters or it will not update
>
> fixed crash with [...]

Oh, I see. Thank you very much!