KlipperScreen/KlipperScreen #606: Bed level screen - 4 screws displayed instead of 3

rubberfistfight, 2022-04-29:
So I've got my screw locations correctly defined in printer.cfg, works a treat if I hit "screws_tilt_adjust" either in the web interface OR via klipperscreen...

But klipperscreen displays 4 corner screws, with co-ordinates I've never assigned and don't know where are sourced from. Instead of back left, back right and front center, it just has one in each corner of a rectangle, none of which correspond to a bed corner.

Is 4 the minimum it'll work with for that? Or am I overlooking something obvious?

rubberfistfight, 2022-04-30:
> 4 is the minimum at the moment

Well that's no good.

I see there was a similar issue raised some time ago that resulted in 6 and 8 screw positions being available. 3-screw beds are common (and are technically superior since you only need 3 points to define a plane and 4 screws is technically overconstrained) so I think this functionality should be added sooner rather than later.

rubberfistfight, 2022-07-15:
Any movement on this?

alfrix, 2022-07-19:
No, i found that the panel has some fundamental issues that need to be addressed before adding new functionality