KlipperScreen/KlipperScreen #608: big peview image

tonywright, 2022-05-04:
Hi, since the last update the preview image is showing so large that it only manages to display quarter of the image and all the rest of the screen is pushed out of view(can't see any buttons or status). The only change I have recently made is to set "#~# screen_blanking = off" changing this back to 99 has made no difference. 
I set the preview image in Cura to 64x64 but even printing a previous working file still shows the very large preview. Is there anything in the last update that might have caused this?
The current version I have is v0.2.2-3-g35381be
Is there a way to roll the version back to a known working version?
I am using a Amazon Fire tablet over Xserver but nothing has changed on the tablet as far as I can see.

Thanks

alfrix, 2022-05-04:
i've fixed it, but i will recommend you to use a bigger size, 64x64 is too small, i've just tested cura2moonraker with ufp and it defaults to 300x300, recommended for fluidd is 300x300
ks needs at least 200x200 on a 480x320 screen to look sharp, and more on higher res displays

tonywright, 2022-05-04:
Thanks for fixing it, I use Cura as my slicer with an add-in for the preview which defaults to 16x16, I have changed it to 200x200 but it makes for some much bigger g-code files :)