KlipperScreen/KlipperScreen #613: Cannot open virtual Console 2

Otterman88, 2022-05-08:
I get this error when running: `journalctl -xe -u KlipperScreen`
[image]
Run `cat /etc/X11/Xwrapper.config`
This should have the line `allowed_users=anybody` in it

Run `cat /etc/group | grep tty`
If your username is not listed under that line, you need to add it with the following command:

`usermod  -a -G tty pi` (if your username is not 'pi' change 'pi' to your username)

If it's still failing, try: `sudo apt install `xserver-xorg-legacy`

As a last resort add `needs_root_rights=yes` to `/etc/X11/Xwrapper.config`

When I try all of these under `cat /etc/group | grep tty` it says access denied.
Any idea what is happening?

Otterman88, 2022-05-09:
It also shows the console instead of klipperscreen.

alfrix, 2022-05-10:
`sudo usermod -a -G tty pi`
reboot
if it's still failing then
`sudo nano /etc/X11/Xwrapper.config`
add a new line with `needs_root_rights=yes`
save and reboot

Otterman88, 2022-05-13:
That worked. Thanks!

closed 2022-06-03 by alfrix