KlipperScreen/KlipperScreen #627: Unable to load panel temperature

p-v-n, 2022-05-28:
When clicking on "Temperature" tile on main screen I get "Unable to load panel temperature" error message.
I have a dual extruder with shared heater/nozzle setup, so it looks like KlipperScreen does not handle this properly.
When I edit printer.cfg to remove the second extruder, everything works.

I'm attaching a zip with the following files:

KlipperScreen.log - KlipperScreen log file immediately after reproducing the issue
printer_broken.cfg - printer config that causes KlipperScreen to not being able to load temp panel
printer_works.cfg - printer config with removed 2nd extruder. KlipperScreen works fine with this one.

[KlipperScreen.zip](https://github.com/jordanruthe/KlipperScreen/files/8792271/KlipperScreen.zip)

alfrix, 2022-05-31:
shared heater is deprecated in main klipper, but i will take a look

p-v-n, 2022-05-31:
> shared heater is deprecated in main klipper, but i will take a look

Thanks! Didn't realize this is being deprecated as of Feb 2022.
FWIW, I have reconfigured my printer with what is suggested as a replacement, and everything is working.

alfrix, 2022-06-15:
well it does conflict with the deprecated config, but it shouldn't matter