KlipperScreen/KlipperScreen #640: [BUG] PyGObject dependencies

mralko99, 2022-06-08:
Hi I noticed that updating to the latest version bricked KlipperScreen.

Following the troubleshooting guide I done:
`journalctl -xe -u KlipperScreen`

The full log is attached [here](https://github.com/jordanruthe/KlipperScreen/files/8861482/journal.log) but the main thing is that:
`module 'gi' has no attribute 'require_version'`

So I unistalled KlipperScreen with Uninstall.sh and tried to reinstall, the full log is [here](https://github.com/jordanruthe/KlipperScreen/files/8861567/installation.log):
The problem is with the installation of PyGObject, it says BadZipfile, I understand the problem but I don't understand how to solve it.

Can someone help me to fix this problem.

alfrix, 2022-06-08:
Sounds like a corrupted download try:
`pip cache purge`
and reinstall

mralko99, 2022-06-11:
Oh thanks, I resolved the issue, also if it was not easy where to pick pip since it was in the virtual-env folder, it could be useful to catch the error of a broken package, I resolved modifying [update-python-venv.sh] https://github.com/jordanruthe/KlipperScreen/files/8883982/update-python-venv.txt)

closed 2022-06-11 by mralko99