KlipperScreen/KlipperScreen #643: [BUG] Job status panel doesn't show up automatically

p-v-n, 2022-06-13:
I believe, based on documentation, that no special configuration necessary to have job status panel activated automatically at the start of the print, however, main menu is being displayed even during print.

**To Reproduce**
Steps to reproduce the behavior:
1. Start print job

**Expected behavior**
Klipperscreen automatically switches to job status panel.

**Logfile**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/8889610/KlipperScreen.log)

Note: Log file was retrieved when first layer started to print.

p-v-n, 2022-06-13:
I just noticed in the Troubleshooting section that Octoprint is not supported. This is what I'm using.
I checked if I can reproduce the issue when starting a print job from Klipperscreen directly and everything is working as expected.
I'm closing this bug.

closed 2022-06-13 by p-v-n

kyngs, 2022-06-18:
I am experiencing the same issue even though I have Klipper. Sometimes it shows up, sometimes not.
Perhaps we should reopen this?