KlipperScreen/KlipperScreen #644: [BUG] No KlipperScreen on Official Pi 7" touchscreen. X server "no screen detected"

water-rhatt, 2022-06-13:
**Describe the bug**
KlipperScreen not loading. Presented only with console login screen.
No `/tmp/KlipperScreen.log` file
Output of `journalctl -xe -u KlipperScreen` includes
```
Jun 13 10:22:18 3dprinter xinit[911]: (EE)
Jun 13 10:22:18 3dprinter xinit[911]: Fatal server error:
Jun 13 10:22:18 3dprinter xinit[911]: (EE) no screens found(EE)
Jun 13 10:22:18 3dprinter xinit[911]: (EE)
Jun 13 10:22:18 3dprinter xinit[911]: Please consult the The X.Org Foundation support
Jun 13 10:22:18 3dprinter xinit[911]:          at http://wiki.x.org
Jun 13 10:22:18 3dprinter xinit[911]:  for help.
Jun 13 10:22:18 3dprinter xinit[911]: (EE) Please also check the log file at "/var/log/Xorg.0.log" for addit>
Jun 13 10:22:18 3dprinter xinit[911]: (EE)
Jun 13 10:22:18 3dprinter xinit[911]: (EE) Server terminated with error (1). Closing log file.
lines 1935-1957/1957 (END)
```

KlipperScreen installed June 13, 2022 with KIAUH. No installation errors reported.

**To Reproduce**
Install KlipperScreen
reboot

**Expected behavior**
KlipperScreen to display on my touchscreen following reboot

water-rhatt, 2022-06-13:
Further details:

If I boot into Pi desktop, rather than "no screens" I get:

```
Jun 13 12:42:54 3dprinter xinit[1818]: (EE)
Jun 13 12:42:54 3dprinter xinit[1818]: Fatal server error:
Jun 13 12:42:54 3dprinter xinit[1818]: (EE) Server is already active for display 0
Jun 13 12:42:54 3dprinter xinit[1818]:         If this server is no longer running, remove /tmp/.X0-lock
Jun 13 12:42:54 3dprinter xinit[1818]:         and start again.
Jun 13 12:42:54 3dprinter xinit[1818]: (EE)
Jun 13 12:42:54 3dprinter xinit[1818]: (EE)
Jun 13 12:42:54 3dprinter xinit[1818]: Please consult the The X.Org Foundation support
Jun 13 12:42:54 3dprinter xinit[1818]:          at http://wiki.x.org
Jun 13 12:42:54 3dprinter xinit[1818]:  for help.
Jun 13 12:42:54 3dprinter xinit[1818]: (EE)
```
I should note that the Touchscreen displays and functions fine in Raspberry Pi Desktop

alfrix, 2022-06-14:
No screens found is usually bad cable or missing driver (usually xserver-xorg-video-fbturbo), 

Server is already active for display x
happens if another xserver is already running, KlipperScreen currently doesn't work properly as a window, so it's not supported to be run inside another UI

closed 2022-06-14 by alfrix