KlipperScreen/KlipperScreen #645: Fails to restart from within KlipperScreen, successful elsewhere

xcasxcursex, 2022-06-14:
**Describe the bug**
When making changes to KlipperScreen's configuration, when a change is made, the user is prompted and the service restarted. Or at least that's how it is meant to go, except the restart is called with sudo, which fails because there is no interactive prompt to enter the password.

Fortunately, the call to sudo was superfluous and simply removing it fixes this. I won't bother with a PR for a single-word change, so here's the link:
https://github.com/jordanruthe/KlipperScreen/blob/eedf5448a0e6540d7eb75385f4c5c72d75b41040/screen.py#L440

**To Reproduce**
Steps to reproduce the behavior:
1. Install KlipperScreen from scratch
2. Change stuff in configuration
3. Note that you are warned about a restart that never happens
4. Observe errors about sudo in the logs
5. Just remove 'sudo' from that line
6. enjoy

**Expected behavior**
Restarting appropriately
I noticed [a seemingly related script](https://github.com/jordanruthe/KlipperScreen/blob/master/scripts/sudo_fix.sh) to make the user get sudo perms without a password, but it's not needed.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**```

Jun 14 13:06:08 Klipper x11vnc[20010]: 2022-06-14 13:06:08,162 [screen_panel.py:on_dropdown_change()] - [main] font_size changed to medium
Jun 14 13:06:08 Klipper x11vnc[20010]: 2022-06-14 13:06:08,169 [screen.py:restart_warning()] - Showing restart warning because: medium
Jun 14 13:06:08 Klipper x11vnc[20010]: 2022-06-14 13:06:08,190 [screen.py:_websocket_callback()] - Filelist changed: {
Jun 14 13:06:08 Klipper x11vnc[20010]:   "action": "modify_file",
Jun 14 13:06:08 Klipper x11vnc[20010]:   "item": {
Jun 14 13:06:08 Klipper x11vnc[20010]:     "modified": 1655175968.1649232,
Jun 14 13:06:08 Klipper x11vnc[20010]:     "size": 658,
Jun 14 13:06:08 Klipper x11vnc[20010]:     "permissions": "rw",
Jun 14 13:06:08 Klipper x11vnc[20010]:     "path": "KlipperScreen.conf",
Jun 14 13:06:08 Klipper x11vnc[20010]:     "root": "config"
Jun 14 13:06:08 Klipper x11vnc[20010]:   }
Jun 14 13:06:08 Klipper x11vnc[20010]: }
Jun 14 13:06:09 Klipper x11vnc[20010]: 2022-06-14 13:06:09,403 [screen.py:restart_ks()] - Restarting
Jun 14 13:06:09 Klipper x11vnc[20089]: sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper
Jun 14 13:06:09 Klipper sudo[20089]: pam_unix(sudo:auth): conversation failed
Jun 14 13:06:09 Klipper sudo[20089]: pam_unix(sudo:auth): auth could not identify password for [MY_USERNAME]
```

alfrix, 2022-06-15:
it always worked, the command should be the same that moonraker uses
https://moonraker.readthedocs.io/en/latest/web_api/#restart-a-system-service

xcasxcursex, 2022-06-15:
Right, and if I restart from mainsail it works, and the command is the same, but the processes apparently have different perms, because it isn't working from within KS.

This isn't a Pi with an image, it's a PC with KIAUH to install, and there's no physical display - so it's pretty nonstandard. I should assume at this point that there's something specific missing from my setup, when compared to a working image.

By what mechanism is it intended to function? I mean, sudo requires user interaction unless there is a process in place to bypass that, so what is that process? That script might do the trick, but it doesn't run by default so I didn't want to just run it blindly.

xcasxcursex, 2022-06-15:
> I didn't want to just run it blindly.

Trying to get less blind on this one...

So as far as I can tell, the default approach to this was that the image had no password required for login. The alternative was to do the usual sudoers NOPASSWD thing, which is no longer needed because moonraker and the frontends use PolicyKit... But KlipperScreen doesn't, so it still needs a fix like that one. Does that sound about right? I'm working backwards from that fix script, to the moonraker one it is based upon, and their documentation of it, but they have the polkit perms applied to a group, and their service runs with that as a supplementary group, and KlipperScreen doesn't, so it seems like KS isn't using that feature.

So anyway I decided I'm less blind now, so I ran the script. Works as expected. There is a typo though: [Line 53](https://github.com/jordanruthe/KlipperScreen/blob/eedf5448a0e6540d7eb75385f4c5c72d75b41040/scripts/sudo_fix.sh#L53) says "IWCONFGI" instead of "IWCONFIG"

Should I talk to our friendly neighbourhood KIAUH people and see if this should be automated, or is this not actually intended to be used and I'm just being a cowboy?

alfrix, 2022-06-16:
> because moonraker and the frontends use PolicyKit... 

that is relatively new

on raspbian you will find /etc/sudoers.d/010_pi-nopasswd
```
pi ALL=(ALL) NOPASSWD: ALL
```
which should explain the difference in a regular pi and your setup, and why usually it's not an issue

i'm thinking of just sending moonraker the command to restart klipperscreen, so moonraker can restart user associated services with the `managed_services` section of moonraker update_manager instead of using sudo systemctl, this is what the system panel does already

xcasxcursex, 2022-06-17:
Thanks mate I'll take it to KIAUH from here, appreciate the help!

closed 2022-06-17 by xcasxcursex

xcasxcursex, 2022-06-17:
> There is a typo though: [Line 53](https://github.com/jordanruthe/KlipperScreen/blob/eedf5448a0e6540d7eb75385f4c5c72d75b41040/scripts/sudo_fix.sh#L53) says "IWCONFGI" instead of "IWCONFIG"

Just so this doesn't fall through the cracks :) I'll file a PR for this if you like?

PR #656 (2022-07-03): Use moonraker to restart KS
fix #645
related to #650