KlipperScreen/KlipperScreen #650: [FR] Polkit for sudo

xcasxcursex, 2022-06-17:
**Is your feature request related to a problem? Please describe.**
KlipperScreen needs sudo_fix.sh

**Describe the solution you'd like**
Use PolKit like the moonraker backend.

**Describe alternatives you've considered**
Automating sudo_fix.sh execution through KIAUH

**Additional context**
As per our discussion earlier, a script needs to be run to make KlipperScreen function on most linux builds, which tripped me up, and I expect I won't be the last.
I took this to KIAUH and filed [an FR](https://github.com/th33xitus/kiauh/issues/212) but the author has a reasonable caution toward automating such a thing, amplified by the fact that moonraker already provides an API to restart KlipperScreen without modifying sudoers.

An easy way to deal with this would be to just restart via moonraker, but I can see some functionality in being able to restart KS without a connection to a backend. 

This is why I figured that it would make sense to either leverage moonraker's permissions (meaning, alter the KS service unit file to include the supplementary group, and call the service restart via dbus) or to implement a similar mechanism (meaning, borrow their code).... or something else you have in mind :)

alfrix, 2022-06-19:
> An easy way to deal with this would be to just restart via moonraker

and probably the best since KS is useless without moonraker, the issue that i see is that people are restarting another service with this function and i will break their configs if i change this because moonraker has a whitelist

> This is why I figured that it would make sense to either leverage moonraker's permissions (meaning, alter the KS service unit file to include the supplementary group, and call the service restart [...]

at the moment i don't know how to implement this, and probably this is too complex for this issue

i'm thinking in making the android users modify the service file instead of creating another one, and add a warning of deprecation of the service setting, then i can just restart KS via moonraker

xcasxcursex, 2022-06-19:
TL;DR at the bottom.

> i will break their configs if i change this because moonraker has a whitelist

Honestly, if people have superuser rights by accident, flowing over from where they should be to somewhere else, and they don't know it... you might be doing them a favour by ~~breaking~~ fixing it for them :) 

> i'm thinking in making the android users modify the service file instead of creating another one, and add a warning of deprecation of the service setting, then i can just restart KS via moonraker

Well that was a rabbit hole for me. I had an idea, but I wanted to make sure it works, so tonight I've had KlipperScreen running on alllll of the things (KS works in 2k on windows over wifi, did you know? Ever seen it on an android TV? :D ). 

One way to deal with this would be to leverage the server argument to xinit, and EnvironmentFile option in systemd. For example, in the service unit file, append `ExecStart` with `-- $KS_XSERVER` and add `EnvironmentFile=KS_DIR/KlipperScreen.Xserver` (Yep, that's it for you. Easy!)

This enables users to have the command to launch the X server in a separate file. If the environment file does not exist or is empty or doesn't set the correct variable, the service uses default (X :0) just as it does now. If the user sets the KS_XSERVER variable in the EnvironmentFile, they will specify an alternative server binary or display, and its appropriate options. 

The alternative server can just be a binary X server like `KS_SERVER="/usr/bin/Xvfb :69 -screen 0 1408x792x24"`
or it can be a script like that used by the common android xsdl approach `KS_SERVER="/home/username/launch_KS.sh :69"`
or even just provide options to the default `KS_SERVER=":42"`

In my POC this works with TTY switching, USB android debugging, wifi android debugging, xvfb, xrdp, x11vnc, you name it. Real screens or virtual, first screen or not, local or remote, they all work like this, with no changes to the service unit file.

The one catch here is that the existing android script needs a small rework so that it stays active, because as far as xinit is concerned, that script *is* the X server, and if the script exits, xinit assumes X exited, and kills KS (even though it would be able to connect just fine). I made this modification by re-using the existing code to poll if the server is ready, and exit the script when that fails, so xinit still cleans up after everything if connection is lost to the android. 

It occurred to me that if you wanted to make it low-impact for the existing android xsdl users, rather than editing their script directly as I did, I could write a small wrapper which would launch the script they use now (lauch_klipperscreen.sh) and then poll its X instance as above. Then they can be given a one-liner to populate the EnvironmentFile appropriately and they're back to normal. The same script also works nicely with other remote X servers like vcxsrv.

I've got all this hashed out on my machine at the moment if you're interested I can paste code in here or make a PR or whatever suits. No pressure, I did it just for fun.

TL;DR:
Add one line and append one option to KlipperScreen.service. No change to existing behaviour with these changes alone. That's all you have to do.
Users can add one line to a new file, to run their own server (eg android xsdl).
Maybe you'd like to include a helper script which I will provide, which will ease the transition for existing users.

alfrix, 2022-06-20:
> so tonight I've had KlipperScreen running on alllll of the things (KS works in 2k on windows over wifi, did you know? Ever seen it on an android TV? :D ).

this one surprised me
[image]

> make a PR

ok :D

xcasxcursex, 2022-06-21:
Sorry to bother you with this but I thought I should make sure I'm headed in the right direction. TL;DR at the bottom

As part of doing this I'm trying to make super-sure it's going to 'just work' for everybody, and so far it does excepting one particular approach which seems to be fairly popular, and my plan above does not account for - That is to not start the X server on the host at all, but start it manually elsewhere, eg android on wifi, then launch KlipperScreen with the DISPLAY environment var set. 

I thought I would be able to work around this without further changes and I will spare you the gory details, except to say that I knew it was truly a rabbit hole when I found myself reading xinit sources 😆 

Given that the wifi android is increasingly common (and honestly is pretty great) and that this approach could be useful in other situations (windows PC, android TV, etc) and I'm here anyway, I'm inclined to try and accommodate loading the client only, but that means having the service call a script to launch klipperscreen, and I'm not sure that's what you'd want.

So TL;DR, should I make a start script for this rather than calling xinit directly, or try to keep changes to the absolute minimum?

xcasxcursex, 2022-06-22:
I spent 30 hours testing and 3 hours repairing the damage from trying out github desktop, for 30 minutes worth of code, but whatever, there 'tis. 

The concept is super simple, rather than calling xinit directly within the service file, we pass xinit's params out to a script to start xinit. 
That script offers end-users an opportunity to override the default behaviour and run a custom script (which doesn't exist in the repo so won't break updates) simply by creating the file and making it executable.

A wrapper is included which will launch the existing script launch_klipperscreen.sh (conveniently typo'd for easily isolation!) and then monitor the server by polling sparsely. This ensures the process stays resident just like a real X server would, and thus the service doesn't attempt to restart when it sees the server process has exited. A one-liner will create the override to run that wrapper:
 `echo "scripts/xsdl_wrapper.sh" > ~/KlipperScreen/scripts/launch_KlipperScreen.sh; chmod +x ~/KlipperScreen/scripts/launch_KlipperScreen.sh`

A few more gory details than my usual, if you feel like TL;DR just back away now.

There are a few design choices that are likely not immediately apparent, but the one I'm most expecting you to go "Why tf did he do that?" is passing client paths as environment vars, which was done primarily in order to avoid touching the installer (otherwise, the substitutions performed by sed to the service file, would have to happen in at least two files), but also means that the user-specific (eg /home/pi) and future-version-specific (eg screen.py) parts of the command are carried from the installer via the service unit file to the script, so users won't have to update their scripts to work through klipperscreen updates, and will be able to simply copy them between different usernames or machines. The usual `lauch_klipperscreen.sh` script gets around this by being in the working directory and not specifying a path at all. Passing the command allows us to isolate the scripts in the script directory and use the full path passed as a var. It also turns out to make it nice and easy to write your own overrides.

One thing I'm really not glued to are the file names and paths. I just tried to follow the existing working directory set in the service file, and allow for the `lauch_klipperscreen.sh` being very much hardcoded where it is, but I don't mind if they get rearranged or renamed. Especially the override script, `launch_KlipperScreen.sh` was chosen purely because it will be familiar (it's the name everyone uses, just with the typos fixed), but it doesn't rally match the existing naming schema, so I wasn't totally cool with it. 

I also have a script here which I haven't committed, which basically acts like xinit but parses full display strings including hosts, and stays up for remote hosts (in a similar fashion to the helper script that's included). I'm kinda inclined to include it, in order to have a facility there to avoid people feeling the need to mess with the .service file again, but I left it out to keep this to the bare minimum to get it working for those two use cases of stock xinit and android xsdl. Maybe one for later?

FWIW, since I'm toying with services and superuser goes there, I did give some light consideration to security. I didn't sweat it too much because I gather it's not a heavy concern with Klipper in general, but the service runs the scripts as the user not as root, the files and paths of which are all under control of the repo and update manager, and importantly, systemd signals (so when the service stops so do their scripts). I considered sourcing the user-supplied script rather than executing it, in order to simplify mods for end-users by skipping the step of applying execution perms to the file, but it permitted the user to interrupt the script called by systemd, so I decided against it. Even if they do manage to break execution, at that point the only thing left is the return code, so it's no issue.

I think that covers off the important parts. I don't remember the last time I put that much thought into that few lines of code :)

alfrix, 2022-06-25:
> avoid touching the installer

the installer needs to be modified to place the new service unit over the old one.

there are 3 main guides that people use for android:
1. https://klipper.discourse.group/t/how-to-klipperscreen-on-android-smart-phones/1196
2. https://www.angelrojasjr.com/2021/10/30/using-an-android-device-with-klipperscreen/?fbclid=IwAR3nre37c3Yhw03n91bnxuhmY03RY5UCz20_KVfLcLUlwp4GXB_CUaTAbIA
3. https://github.com/naruhaxor/AndroidKlipperScreen

all of them use `launch_klipperscreen.sh` in `~/KlipperScreen/`

so that folder and capitalization would be preferred, to help most users to migrate seamlessly 

guide number 3 uses `KlippyScreenAndroid.service` that probably should be disabled by the installer to move those users to the new method

another issue is the need to re-run the installer, which doesn't happen with an update

a bit off-topic, i'm tempted to write a guide for android and include it in the docs

xcasxcursex, 2022-06-27:
> all of them use `launch_klipperscreen.sh` in `~/KlipperScreen/`

The first one uses 'lauch_klipperscreen', no 'n', I wasn't joking about the typo :)

> so that folder and capitalization would be preferred, to help most users to migrate seamlessly

If only.... the trouble is that the current systemd service expects the process to start and stay resident, like X does, but the existing `lauch_klipperscreen.sh` script starts and exits, so the service sees the exit, and restarts in a loop. 

If we just simply point the existing service at their script, it won't work, so the approach is.... 

- One script so we can support custom scripts, 
- a custom script (doesn't exist, must be created) so we can not have users edit files from the repo, 
- a wrapper script so the original works like the service expects, 
- and then the original....

So the custom script can't have the same path as the original.

I'm not glued to doing it that way, but I figured you had configured the service to keep the stock klipperscreen (ie, xinit calls X on :0 and the client) resilient to failures by detecting a server exit and restarting itself, and you'd want to keep it that way. It does make things tricky in those setups where the X server isn't a running, local process.

> a bit off-topic, i'm tempted to write a guide for android and include it in the docs

I've done that a bunch of times now haha I can definitely help out if you like?

alfrix, 2022-06-27:
i've merged your PR into another branch until the feature is ready, i want to migrate the users without breaking their systems.

> I've done that a bunch of times now haha I can definitely help out if you like?

i'll do a base document and you can PR later, any help is appreciated

xcasxcursex, 2022-06-28:
> i want to migrate the users without breaking their systems.

Right. I've been giving this a _crazy_ amount of thought. I think it's probably obvious by now that I've been going to some lengths to be as low-impact as possible, and change as little of this repo, and the end-users setup, as possible, but.....

The real problem here is that the target audience doesn't have one particular known setup from which to migrate. There are those three links above, and each of them has a few variations, individuals have changed spelling on file names, etc etc.

The approach so far has been to try and allow for any of these unknown setups to be easily integrated with KS. But I'm not sure that's the right way to go. 

If you really want this to go smoothly, I think the best option is to include your own script to launch klipper on android (based on the existing scripts, of course). If you allow for a phone connected by USB and using adb, and a wifi-connected phone with a user-supplied IP address, you're going to cover the vast majority of the existing implementations - far more than I can imagine working using something like I've written above and 'mysteryscript.sh' on the user's machine. It'll also make troubleshooting any failures far more straightforward because you don't have to reverse engineer their existing unknown setup to figure out where the migration went wrong.

I've been really hesitant to come right out and say this because I wasn't confident that you'd be keen on the idea, but, the more I think about it, the more ways I can imagine for a migration of existing scripts to go wrong, and so I think I should suggest it.

Of course, I'm volunteering any time you need spent on it, including hanging around for helpdesk duties after the migration.

alfrix, 2022-06-29:
i generally agree, the only thing that will 'break' is the restart method of those who followed the tutorial number 3
i will provide the general guide with the new method and the adb script ready to go
and a solution for users of the affected method in the docs.

The current guide:
https://github.com/jordanruthe/KlipperScreen/blob/launcher/docs/Android.md
https://github.com/jordanruthe/KlipperScreen/blob/launcher/scripts/android-adb.sh

if you want to PR target the `launcher` branch
i'll probably going to merge everything this weekend

closed 2022-07-03 by alfrix

xcasxcursex, 2022-07-07:
Sorry I've been slow to respond this week, IRL disaster happened and my printer required immediate attention. It's going to be dismantled by next week and I'll be full-time looking at this for you. I'll update over in that repo.

> the only thing that will 'break' is the restart method of those who followed the tutorial number 3

Hmmm hahah I wrote a long reply with details about edge cases but I'll spare you the tedium. All those problems are easy solved now that you have a script that can be designed to behave as xinit does.

PR #651 (2022-07-03): 0.2.4
close #414 performance of the graph will never be very smooth with matplotlib and it will be terrible through xsdl or resistive touchscreens that refresh at 20fps, but with a native screen that has a good refresh speed it's acceptable after removing the background.

close #558 (use .action_bar{} and .content{} in a custom theme)

close #650 not going to do polkit but the launcher that was talked about

close #619 caused by the background [636e277](https://github.com/jordanruthe/KlipperScreen/pull/651/commits/636e277144f2e145102f61753f91c52576970163)