KlipperScreen/KlipperScreen #662: Not able to install KlipperScreen [BUG]

basha318i, 2022-06-27:
I'm trying to install KlipperScreen on Armbian Buullseye; I was able to get it to work before, but now I'm not. I tried KIAUH & installing KlipperScreen manually, it's always failing for some reason.

**To Reproduce**
Steps to reproduce the behavior:
1. Start with a fresh install of Armbian
2. sudo apt update/upgrade
3. git KIAUH
4. Install Klipper, Mainsail
5. install KlipperScreen; you would get to this point:
Collecting matplotlib==3.5.0
  Using cached matplotlib-3.5.0.tar.gz (35.0 MB)
  Preparing metadata (setup.py) ... done
  WARNING: Requested matplotlib==3.5.0 from https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa, but installing version 0.0
WARNING: Discarding https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa (from https://pypi.org/simple/matplotlib/) (requires-python:>=3.7). Requested matplotlib==3.5.0 from https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa has inconsistent version: filename has '3.5.0', but metadata has '0.0'
ERROR: Could not find a version that satisfies the requirement matplotlib==3.5.0 (from versions: 0.86, 0.86.1, 0.86.2, 0.91.0, 0.91.1, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1rc1, 1.4.1, 1.4.2, 1.4.3, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 2.0.0b1, 2.0.0b2, 2.0.0b3, 2.0.0b4, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.0.1, 2.0.2, 2.1.0rc1, 2.1.0, 2.1.1, 2.1.2, 2.2.0rc1, 2.2.0, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 3.0.0rc2, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0rc1, 3.1.0rc2, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.2.0rc1, 3.2.0rc3, 3.2.0, 3.2.1, 3.2.2, 3.3.0rc1, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.4.0rc1, 3.4.0rc2, 3.4.0rc3, 3.4.0, 3.4.1, 3.4.2, 3.4.3, 3.5.0b1, 3.5.0rc1, 3.5.0, 3.5.1, 3.5.2)
ERROR: No matching distribution found for matplotlib==3.5.0
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.

**Expected behavior**
it should install normally

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
/tmp/KlipperScreen.log does not exist

alfrix, 2022-06-27:
not really a bug, it's a bad download
do:
```
source ~/.KlipperScreen-env/bin/activate
pip cache purge
deactivate
```

basha318i, 2022-06-27:
@alfrix thanks a lot for your quick response, I have tried several times, always failing with the same error. Even with a fresh image of Armbian.

basha318i, 2022-06-27:
same error again:

Installing collected packages: MarkupSafe, jinja2
Successfully installed MarkupSafe-2.1.1 jinja2-3.1.2
Collecting matplotlib==3.5.0
  Downloading matplotlib-3.5.0.tar.gz (35.0 MB)
     |████████████████████████████████| 35.0 MB 10 kB/s             
  Preparing metadata (setup.py) ... done
  WARNING: Requested matplotlib==3.5.0 from https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa, but installing version 0.0
WARNING: Discarding https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa (from https://pypi.org/simple/matplotlib/) (requires-python:>=3.7). Requested matplotlib==3.5.0 from https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa has inconsistent version: filename has '3.5.0', but metadata has '0.0'
ERROR: Could not find a version that satisfies the requirement matplotlib==3.5.0 (from versions: 0.86, 0.86.1, 0.86.2, 0.91.0, 0.91.1, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1rc1, 1.4.1, 1.4.2, 1.4.3, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 2.0.0b1, 2.0.0b2, 2.0.0b3, 2.0.0b4, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.0.1, 2.0.2, 2.1.0rc1, 2.1.0, 2.1.1, 2.1.2, 2.2.0rc1, 2.2.0, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 3.0.0rc2, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0rc1, 3.1.0rc2, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.2.0rc1, 3.2.0rc3, 3.2.0, 3.2.1, 3.2.2, 3.3.0rc1, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.4.0rc1, 3.4.0rc2, 3.4.0rc3, 3.4.0, 3.4.1, 3.4.2, 3.4.3, 3.5.0b1, 3.5.0rc1, 3.5.0, 3.5.1, 3.5.2)
ERROR: No matching distribution found for matplotlib==3.5.0
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.

basha318i, 2022-06-27:
it is indeed pip 21.3.1. I will try to upgrade & keep you posted.. thanks a lot.

alfrix, 2022-06-27:
:facepalm: i specifically said to not upgrade

basha318i, 2022-06-27:
same error even after updating pip:

Creating virtual environment
Requirement already satisfied: numpy==1.21.4 in /home/pi/.KlipperScreen-env/lib/python3.9/site-packages (1.21.4)
Requirement already satisfied: jinja2==3.1.2 in /home/pi/.KlipperScreen-env/lib/python3.9/site-packages (3.1.2)
Requirement already satisfied: MarkupSafe>=2.0 in /home/pi/.KlipperScreen-env/lib/python3.9/site-packages (from jinja2==3.1.2) (2.1.1)
Collecting matplotlib==3.5.0
  Using cached matplotlib-3.5.0.tar.gz (35.0 MB)
  Preparing metadata (setup.py) ... done
  WARNING: Requested matplotlib==3.5.0 from https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa, but installing version 0.0
Discarding https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa (from https://pypi.org/simple/matplotlib/) (requires-python:>=3.7): Requested matplotlib==3.5.0 from https://files.pythonhosted.org/packages/78/cf/96af81ef2c15682a020a992cb0f1291bb8ee0b3c296f5cd7978348a72993/matplotlib-3.5.0.tar.gz#sha256=38892a254420d95594285077276162a5e9e9c30b6da08bdc2a4d53331ad9a6fa has inconsistent version: filename has '3.5.0', but metadata has '0.0'
ERROR: Could not find a version that satisfies the requirement matplotlib==3.5.0 (from versions: 0.86, 0.86.1, 0.86.2, 0.91.0, 0.91.1, 1.0.1, 1.1.0, 1.1.1, 1.2.0, 1.2.1, 1.3.0, 1.3.1, 1.4.0, 1.4.1rc1, 1.4.1, 1.4.2, 1.4.3, 1.5.0, 1.5.1, 1.5.2, 1.5.3, 2.0.0b1, 2.0.0b2, 2.0.0b3, 2.0.0b4, 2.0.0rc1, 2.0.0rc2, 2.0.0, 2.0.1, 2.0.2, 2.1.0rc1, 2.1.0, 2.1.1, 2.1.2, 2.2.0rc1, 2.2.0, 2.2.2, 2.2.3, 2.2.4, 2.2.5, 3.0.0rc2, 3.0.0, 3.0.1, 3.0.2, 3.0.3, 3.1.0rc1, 3.1.0rc2, 3.1.0, 3.1.1, 3.1.2, 3.1.3, 3.2.0rc1, 3.2.0rc3, 3.2.0, 3.2.1, 3.2.2, 3.3.0rc1, 3.3.0, 3.3.1, 3.3.2, 3.3.3, 3.3.4, 3.4.0rc1, 3.4.0rc2, 3.4.0rc3, 3.4.0, 3.4.1, 3.4.2, 3.4.3, 3.5.0b1, 3.5.0rc1, 3.5.0, 3.5.1, 3.5.2)
ERROR: No matching distribution found for matplotlib==3.5.0
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.

alfrix, 2022-06-27:
maybe this is an issue with the webhost package, i haven't seen this before.
as workaround on bullseye you can run newer versions, you need to upgrade numpy too
install the requirements manually with
```
source ~/.KlipperScreen-env/bin/activate
pip uninstall numpy
pip install numpy
pip install matplotlib
```
then continue with the install

basha318i, 2022-06-27:
Oh sorry, I didn't realize that!

I tried that, so it kept downloading & discarding versions..

[image]

basha318i, 2022-06-27:
but version matplotlib-3.4.3.tar.gz did the trick.. not sure how would that impact the final product

valeriiyurchenko, 2022-06-28:
i have the same issue. Tried to install 5 times with fresh armbian and debian and ubuntu, and every time result was the same - ERROR: No matching distribution found for matplotlib==3.5.0
When i downloaded matplotlib-3.5.0 and installed manualy it said: Successfully installed matplotlib-0.0

basha318i, 2022-06-28:
@valeriiyurchenko what’s driving me nuts is that I was able to install it several times before using the same method & hardware 🤯

valeriiyurchenko, 2022-06-29:
@basha318i my workaround how to install it:
```
source ~/.KlipperScreen-env/bin/activate
sudo apt-get install python3-matplotlib
./KlipperScreen-install.sh
```

alfrix, 2022-10-16:
recently tested this on armbian and it didn't happen, so i suppose it was fixed in upstream

closed 2022-10-16 by alfrix