KlipperScreen/KlipperScreen #665: [BUG] Raspberry Pi Os Lite 04-04-2022

ETE-Design, 2022-07-03:
With Raspbian Lite OS Lite - 04-04-2022 HyperPixel4 is baked into the OS, so you now only need to enable it in Config.txt

`dtoverlay=vc4-kms-dpi-hyperpixel4`
`dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y`

Problem is that is seems like KlipperScreen dosen't accept that change, at least not the Rotate part :-( Raspberry PI OS Lite Promt is ratated to landscape and the touch also, but when you start KlipperScreen it start in Portrait Mode no matter what - but Touch is still rotated. Can't figure out what to change :-( Could be a great feature to make a rotating setting build into KlipperScreen.

alfrix, 2022-07-03:
This is not a bug of klipperscreen, you need to properly configure you screen
https://github.com/pimoroni/hyperpixel4#persisting-rotation
`Section "Monitor"` is the visual and and CalibrationMatrix the touch
do not pay attention to the lightdm recommendations, lightdm is not used for KS

there are some calibration matrixes in the docs if you need them
https://klipperscreen.readthedocs.io/en/latest/Hardware/?h=rotat#touchscreen-touch-rotation

closed 2022-07-03 by alfrix

ETE-Design, 2022-07-03:
@alfrix This is for the "Old" Versions... Not working on the new one... Look here: https://github.com/pimoroni/hyperpixel4/issues/177

alfrix, 2022-07-03:
no, it's still a valid way to rotate the screen the only thing i see changed are identifiers
from
```
Section "Monitor"
	Identifier "DSI-1"
	Option "Rotate" "left"
EndSection
```
to
```
Section "Monitor"
	Identifier "DPI-1"
	Option "Rotate" "left"
EndSection
```
put the file in /etc/X11/xorg.conf.d/90-monitor.conf
and restart klipperscreen
and the Identifier of the touch part, but the guide on klipperscreen docs should be good enough if you need to change the touch

alfrix, 2022-07-03:
and by the way, rotation on the fly it's not supported at all in KlipperScreen, the server needs to start at the correct orientation