KlipperScreen/KlipperScreen #674: On Screen Keyboard Not Showing

bholdmeyer, 2022-07-16:
The on-screen keyboard has stopped working on my screen in the console panel.  Whenever I tap the input box, the cursor shows up but the keyboard doesn't pop up.

My log file shows this:

2022-07-16 01:47:25,320 [functions.py:logging_exception_handler()] - Uncaught exception <class 'ValueError'>: invalid literal for int() 
with base 10: b''
Traceback:   File "/home/pi/KlipperScreen/screen.py", line 1034, in show_keyboard
    xid = int(p.stdout.readline())
NoneType: None

I have tried uninstalling and removing KlipperScreen and then installing fresh and the issue persists.

alfrix, 2022-07-16:
maybe the dependency wasn't installed, it is impossible to know without looking at the installer output
try `sudo apt install matchbox-keyboard`
the installer should have taken care of it

alfrix, 2022-07-17:
i have worked on built-in OSK that doesn't requiere external dependencies, but itwasn't meged because it still lacks the ability to  move the cursor

bholdmeyer, 2022-07-18:
@alfrix 

That was one of the first things I tried before uninstalling and reinstalling KlipperScreen.  I just ran it again and got the same output as before:

matchbox-keyboard is already the newest version (0.1+svn20080916-12).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

To clarify, the keyboard was working fine when I first installed KlipperScreen, and then suddenly I noticed it was no longer coming up when selecting the text input box on the console screen.  

The only things I changed were the theme and font size and I have tried reverting those to defaults and the keyboard still does not show and I get the same error as stated above.

closed 2022-07-18 by alfrix