KlipperScreen/KlipperScreen #683: Klipper screen stuck 'Connecting to Printer'

drak42, 2022-07-23:
Hi,

Recently updated all components throug Kiahu and now Klipperscreen just has 'Connecting to printer' on the screen.
All functionality is fine if I connect to the Printer through Fluidd or Mainsail. 

I have tried re install both moonraker a Kliiperscreen but still the same issue.

Any Thoughts?

Going to Server info Page does bring back information, all I can find is this in logs

Jul 23 12:24:50 klippi KlipperScreen-start.sh[508]: Traceback:   File "/home/pi/KlipperScreen/screen.py", line 974, in init_printer
Jul 23 12:24:50 klippi KlipperScreen-start.sh[508]:     self.printer.reinit(printer_info['result'], config['result']['status'])
Jul 23 12:24:50 klippi KlipperScreen-start.sh[508]:   File "/home/pi/KlipperScreen/ks_includes/printer.py", line 90, in reinit
Jul 23 12:24:50 klippi KlipperScreen-start.sh[508]:     r['x_count'] = int(r['x_count'])
Jul 23 12:24:50 klippi KlipperScreen-start.sh[508]: NoneType: None

Let me know what log files you may want

TIA
Chris

drak42, 2022-07-24:
Applied further updates and now all working... no idea what changed though

alfrix, 2022-07-28:
it is a very rare failure, i've never seen it, i suppose it was something in your config

closed 2022-07-28 by alfrix