KlipperScreen/KlipperScreen #690: [BUG] I just updated 4 of my printers, and Klipperscreen on one of them doesn't connect properly

simpat1zq, 2022-08-01:
**Describe the bug**
I updated moonraker, and KlipperScreen on 4 printers. For 3 of them, the print was already running, so I'm not sure if they will have the same issue or not. But the 4th one, when I start a print, the print starts just fine, but KlipperScreen just shows me the last directory I was looking at. If I restart KlipperScreen, it doesn't connect and just says "Connecting to Printer". Moonraker seems OK since I'm able to use Fluidd just fine. 

The version is  v0.2.5-2-g16871f9 

I've tried uninstalling and reinstalling KS (via kiauh). I've tried using the localhost IP, and the ethernet IP, and both do the same thing. 

This is what I see in the logs when I start a print:
`2022-08-01 18:46:57,044 [screen.py:_load_panel()] - Loading panel: job_status
2022-08-01 18:46:57,044 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/job_status.py
2022-08-01 18:46:57,129 [job_status.py:set_state()] - Changing job_status state from 'standby' to 'printing'
2022-08-01 18:46:57,131 [job_status.py:update_file_metadata()] - Update Metadata. File: Catan/Organizer/Docksx1.gcode Size: 4996409
2022-08-01 18:46:57,151 [functions.py:logging_exception_handler()] - Uncaught exception <class 'TypeError'>: unsupported operand type(s) for -: 'NoneType' and 'float'
Traceback: ['  File "/home/pi/KlipperScreen/screen.py", line 723, in state_execute\n    callback(prev_state)\n', '  File "/home/pi/KlipperScreen/screen.py", line 777, in state_printing\n    self.printer_printing()\n', '  File "/home/pi/KlipperScreen/screen.py", line 1030, in printer_printing\n    self.show_panel(\'job_status\', "job_status", "Print Status", 2)\n', '  File "/home/pi/KlipperScreen/screen.py", line 323, in show_panel\n    self.panels[panel_name].process_update("notify_status_update", self.printer.get_data())\n', '  File "/home/pi/KlipperScreen/panels/job_status.py", line 653, in process_update\n    self.update_time_left(duration, ps[\'filament_used\'])\n', '  File "/home/pi/KlipperScreen/panels/job_status.py", line 721, in update_time_left\n    self.update_text("time_left", self.format_time(total_duration - duration))\n']
NoneType: None
`

simpat1zq, 2022-08-01:
Quick update. I restarted KlipperScreen on 2 of the other 3 printers and got the same result. Klipperscreen doesn't connect to Moonraker. Is this is a possible bug with the new release? I just now realized that it's only 7 hours old. Please let me know if there is anything you would like for me to try.

simpat1zq, 2022-08-01:
And now I just restart KS on that 4th printer, and it is behaving just fine. That one is a Pi3. The other 3 KS that have the issue are Pi4 and Pi400s(which I believe are basically just Pi4's). I wonder if there is something that is on the Pi4's that it doesn't like. I also just updated the system updates for one of the Pi4, and that didn't help.

alfrix, 2022-08-01:
it sounds like you didn't put `[display_status]` on printer.cfg without the full log idk, i'll try to safeguard the panel more, but the time will not work properly without the required config

simpat1zq, 2022-08-01:
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/9237917/KlipperScreen.log)
Sorry about that. I've included the full log. And I do have [display_status] in there. This setup has been running through many versions of KlipperScreen.

simpat1zq, 2022-08-01:
And now, one of the printers that weren't connecting just finished a print, and it connected. And I started another print and it stayed at the directories screen. I have included the logs that contains all that. Just curious are you running a Pi4?
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/9237936/KlipperScreen.log)

alfrix, 2022-08-01:
if understand correctly the file metadata and stats were not ready when it tried to calculate time

closed 2022-08-01 by alfrix

simpat1zq, 2022-08-01:
I updated all 4 printers, and they all connected right away. They are all printing at this time, but this looks like it fixed the issue. Thanks so much for the quick responses.

alfrix, 2022-08-02:
thank you for taking the time to do a proper report, sorry for the inconvenience

commit c06ca56e (2022-08-01): job_status: fix #690