KlipperScreen/KlipperScreen #691: [BUG]/[FR] Material Darker Theme no longer supports AMOLED Screens as of v0.2.5

ChaosBlades, 2022-08-02:
In the recent update that just released (v0.2.5) the Material Darker theme was updated and no longer results in pure black on OLED/AMOLED screens. It is now dark grey.

I propose two solutions. Either revert the change so that Material Darker theme has pure blacks or add two additional themes to the current list that are variants of the Material themes. You could call them OLED-Dark and OLED-Darker.

This is the screen I am using.
https://www.waveshare.com/wiki/5.5inch_HDMI_AMOLED

ChaosBlades, 2022-08-02:
Looks like on the current update the buttons are pure black and the background is dark grey. It should be the other way around on both variants of the Material themes. Switch those around and I think we are good even on LCDs.

alfrix, 2022-08-02:
"no longer supports AMOLED" it's just not true, you don't need to use all black all the time, material themes never targeted pure black, if you want to do a custom theme please go ahead, it's explained in the docs
https://klipperscreen.readthedocs.io/en/latest/Theming/

closed 2022-08-02 by alfrix

ChaosBlades, 2022-08-02:
It used to be pure black and allowed the background to blend into the bezel on AMOLED screens. Now it does not hence 'no longer supports' as in does not support the best feature of the screen. 

It is disappointing that Material Darker looked fantastic on AMOLED screens out of the box now anyone that uses one need to go out of there way to create a custom theme. The dark grey is barley even indistinguishable from the blackest black an LCD can produce anyways. I just don't see the point in not fixing this. It is not like this screen I am using isn't popular as a KlipperScreen display. It even has a Voron mod. Not to mention all the people that use KlipperScreen on SmartPhones and Tablets where AMOLED is popular.

As far as I see it the latest update has dropped support for AMOLED and I just don't see the point.

ChaosBlades, 2022-08-02:
To anyone that searched and found this you don't need to create a custom theme from scratch. All you need to do is either SSH or use something like FileZilla and go to `/home/pi/KlipperScreen/styles`, enter either material-dark or material-darker, edit the style.css file on line 8 like so.

```
window {
    background-color: #000000;
}
```

I am guessing this will revert on every KlipperScreen update so copy and paste the theme you edited and rename them from material-dark to oled-dark then copy them back to the same directory then restart KlipperScreen and apply the newly created theme from the settings menu.

You can also add in the old material themes I extracted from the previous update and put them into the same folder. I already renamed them old-material-*. Just drop them into that folder and apply the theme.
[old-material-dark.zip](https://github.com/jordanruthe/KlipperScreen/files/9240711/old-material-dark.zip)

ChaosBlades, 2022-09-05:
The latest update changed the screensaver to something other than pure black. This is even worse than the annoyance of non pure black backgrounds as it does not allow the pixels to turn off resulting in premature failure of the display. So I needed to re-edit the themes. Also since last post I have found several other screens that have a non pure black background. So only editing 1 line does not provide the correct result. For now and hopefully the future to make the material themes OLED compatible Find and Replace All references to the following in the style.css file.

Material Dark
Find and Replace All #121212 with #000000 (This will make all backgrounds pure black)
Find and Replace All #090909 with #282828 (This will make the buttons the same color as the scroll bar)

Material Darker
Find and Replace All #121212 with #000000 (This will make all backgrounds pure black)

Here are the themes pre-edited.
[oled-dark|darker.zip](https://github.com/jordanruthe/KlipperScreen/files/9486220/oled-dark.darker.zip)

alfrix, 2022-09-06:
> The latest update changed the screensaver to something other than pure black

that's a bug, thanks for reporting it

>  So I needed to re-edit the themes

as said create a new theme that matches your expectations, if you want a base just copy theme you like into another folder and rename it, so when you change stuff it doesn't invalidate the repo,  just put your theme in another folder (ks needs to be restarted for theme to appear in the list)

it's explained in https://klipperscreen.readthedocs.io/en/latest/Theming/

other people have done similar things, it's normal that you want your own thing, that's why you can create your own like this user:
[image]

_The recommended dark theme surface color is #121212._
https://material.io/design/color/dark-theme.html

ChaosBlades, 2022-09-06:
@alfrix Looks like the fix for the screen saver introduced a new bug. The last leg of the m in am for the time is still visible when the screen saver kicks in.

I am just updating this so that others that want to leverage the unique aspects of OLED displays can while performing as little work as possible. That is why I posted this issue in the first place. I really do not care what theme is added or edited to support OLED. I only singled out the Material Design theme because it had a pure black background at one point. When I first installed KlipperScreen I literally just went through all the themes to find the one with the pure black background.

Now that I read the documentation I see why Google is setting the background to #121212 

> Caution
> On OLED screens, turning pixels on and off can cause a delay when the screen is scrolled, making the pixels blur.

But that issue does not really apply to KlipperScreen since you almost never scroll.