KlipperScreen/KlipperScreen #692: [FR] [Semi-Bug/Unexpected Behaviour] Setting to set if SAVE_CONFIG should be prompted during print

TekkertheChaot, 2022-08-03:
**Is your feature request related to a problem? Please describe.**
As described in the affected commit (https://github.com/jordanruthe/KlipperScreen/commit/fda5589531c0ff7bb4920771ae9a7bb1b5c61ec0#commitcomment-80107877): 
everytime my auto level finishes and the actual print starts, klipperscreen asks me to save the config and restart Klipper while printing.

**Describe the solution you'd like**
I do like the change made with this commit but I'd suggest a switch/setting to decide, if this pop up should be prompted during print / printer state: printing.

**Describe alternatives you've considered**
Alternatively, a global switch to switch back to the old behaviour would be great. 
Or maybe the is a whole another approach to do a bed calibrate that is used for the current print without raising `SAVE_CONFIG`? 

**Additional context**
I do a autobed leveling before each print because my bed and springs seem to be warping with every heating.
I use following gcode as a backwards compatibility with older generated gcode: `[gcode_macro G29] gcode: BED_MESH_CALIBRATE PROFILE=ender3 BED_MESH_PROFILE LOAD=ender3`

alfrix, 2022-08-03:
i do agree that it shouldn't be asking to reboot while printing, imho running g29 on every print it's overkill, you may want to tighten the screws, or some other solution, because it's not normal to need that much calibration

closed 2022-08-03 by alfrix

TekkertheChaot, 2022-08-03:
That was fast, thank you for the quick fix!
You absolutely right, i already got some improved springs and they did help but for some reason there is still too much variance between prints, so unless I find a better solution (I'm still rather new to 3D printing in general) overkill it is right now. 
Again thanks for your feedback!

commit 835803d3 (2022-08-03): screen: do not ask to save unless printer is in standby close #692