KlipperScreen/KlipperScreen #693: [BUG]Save Bed mesh screen shows if print starts with BED_MESH_CALIBRATE in the START_PRINT macro

tikiss, 2022-08-08:
**Describe the bug**
Save Bed mesh screen shown during printing.
Since the last update, I performed I got the save bed mesh screen every time I start a print.
I am using the START_PRINT gcode macro in every print, this macro contains the following which is bed mesh related:
```
  G28 Z ;Home again as Z will have changed after tilt adjustment and bed heating.
  M117 Bed mesh calibration ...
  BED_MESH_CALIBRATE PROFILE=ratos ; Calibrate bed mesh
  BED_MESH_PROFILE LOAD=ratos ; Load previously saved mesh
```
The screen remains there as long as I click "Cancel" or "Save".
If I click "Save" the printer restarts (due to the config being changed) and I have to start the print again.
If I click "Cancel" I just go back to the printing screen and the print end normally.
If nothing is clicked, print goes further but the print screen is not visible but the Save Bed mesh.

**Printer**
I have a RatRig V-Core 3, with Octopus Pro 446 board.
It has a Raspberry Pi 4 built-in running RatOS on it.
SW Versions:
klipper: v0.10.0-546-ga709ba43
KlipperScreen: v0.2.5-11-g0ec2baf
RatOS: v1.2.3-0-g1b1c922
RatOSTheme: v1.2.0-1-g692057f
led_effect: v0.0.5-1-gf12dcdc
moonraker: v0.7.1-617-g0188b81
moonraker-telegram: v1.0.2-222-g4a2751c
mainsail: v2.2.1

**Expected behavior**
Before the SW update (not just KlipperScreen), KlipperScreen showed the Printing screen after the Bed mesh was done.

**Screenshots**
[image]

**Logfile**
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/9282023/KlipperScreen.log)

alfrix, 2022-08-08:
you need to update, already addressed in this commit [0ec2](https://github.com/jordanruthe/KlipperScreen/commit/0ec2baf302d66fa7df28bd3cd58034d35d2ac521)

closed 2022-08-08 by alfrix

tikiss, 2022-08-08:
But I have the version you mentioned installed: v0.2.5-11-g0ec2baf or is it not the same as what you mentioned (0ec2)?

alfrix, 2022-08-08:
sorry the commit is 835803d329e093594f4f9e66fb42ec627fad608a

Dids, 2023-01-27:
This still seems to be an issue? In my custom start print macro, I'm loading the bed mesh and probing a specific area based on the printable object's size, but every time the probing finishes, KlipperScreen is stuck on the save config screen.