KlipperScreen/KlipperScreen #695: Klipperscreen not startin

bulbersator, 2022-08-10:
Hi i installed klipperscreen using KIAUH on RPI zero 2w. 

here's the xorg log
`[   764.814] 
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[   764.816] Build Operating System: linux Raspbian
[   764.817] Current Operating System: Linux Anetka 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l
[   764.817] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 video=HDMI-A-1:800x480M@59 smsc95xx.macaddr=B8:27:EB:8C:8C:5A vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=tty1 root=PARTUUID=0a2ad2aa-02 rootfstype=ext4 fsck.repair=yes rootwait
[   764.818] Build Date: 08 August 2022  07:20:04AM
[   764.819] xorg-server 2:1.20.11-1+rpt1+deb11u2 (https://www.debian.org/support) 
[   764.819] Current version of pixman: 0.40.0
[   764.820] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[   764.820] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[   764.823] (==) Log file: "/home/pi/.local/share/xorg/Xorg.0.log", Time: Wed Aug 10 12:01:48 2022
[   764.825] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[   764.826] (==) No Layout section.  Using the first Screen section.
[   764.826] (==) No screen section available. Using defaults.
[   764.826] (**) |-->Screen "Default Screen Section" (0)
[   764.826] (**) |   |-->Monitor "<default monitor>"
[   764.827] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[   764.827] (**) Option "Debug" "dmabuf_capable"
[   764.827] (==) Automatically adding devices
[   764.827] (==) Automatically enabling devices
[   764.827] (==) Automatically adding GPU devices
[   764.827] (==) Max clients allowed: 256, resource mask: 0x1fffff
[   764.827] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[   764.828] 	Entry deleted from font path.
[   764.828] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[   764.828] 	Entry deleted from font path.
[   764.828] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[   764.828] 	Entry deleted from font path.
[   764.828] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[   764.828] 	Entry deleted from font path.
[   764.828] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[   764.828] 	Entry deleted from font path.
[   764.828] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[   764.828] 	Entry deleted from font path.
[   764.828] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[   764.828] (==) ModulePath set to "/usr/lib/xorg/modules"
[   764.828] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[   764.828] (II) Loader magic: 0x20cf40
[   764.828] (II) Module ABI versions:
[   764.828] 	X.Org ANSI C Emulation: 0.4
[   764.828] 	X.Org Video Driver: 24.1
[   764.828] 	X.Org XInput driver : 24.1
[   764.828] 	X.Org Server Extension : 10.0
[   764.831] (--) using VT number 2

[   764.831] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[   764.834] (II) xfree86: Adding drm device (/dev/dri/card0)
[   764.838] (II) no primary bus or device found
[   764.839] 	falling back to /sys/devices/platform/soc/soc:gpu/drm/card0
[   764.839] (II) LoadModule: "glx"
[   764.839] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[   764.843] (II) Module glx: vendor="X.Org Foundation"
[   764.843] 	compiled for 1.20.11, module version = 1.0.0
[   764.843] 	ABI class: X.Org Server Extension, version 10.0
[   764.843] (==) Matched modesetting as autoconfigured driver 0
[   764.843] (==) Matched fbdev as autoconfigured driver 1
[   764.843] (==) Assigned the driver to the xf86ConfigLayout
[   764.844] (II) LoadModule: "modesetting"
[   764.844] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so
[   764.844] (II) Module modesetting: vendor="X.Org Foundation"
[   764.844] 	compiled for 1.20.11, module version = 1.20.11
[   764.844] 	Module class: X.Org Video Driver
[   764.844] 	ABI class: X.Org Video Driver, version 24.1
[   764.845] (II) LoadModule: "fbdev"
[   764.845] (II) Loading /usr/lib/xorg/modules/drivers/fbdev_drv.so
[   764.845] (II) Module fbdev: vendor="X.Org Foundation"
[   764.845] 	compiled for 1.20.3, module version = 0.5.0
[   764.845] 	Module class: X.Org Video Driver
[   764.845] 	ABI class: X.Org Video Driver, version 24.0
[   764.845] (II) modesetting: Driver for Modesetting Kernel Drivers: kms
[   764.845] (II) FBDEV: driver for framebuffer: fbdev
[   764.845] (EE) 
Fatal server error:
[   764.846] (EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
[   764.846] (EE) 
[   764.846] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   764.846] (EE) Please also check the log file at "/home/pi/.local/share/xorg/Xorg.0.log" for additional information.
[   764.846] (EE) 
[   764.846] (WW) xf86CloseConsole: KDSETMODE failed: Bad file descriptor
[   764.846] (WW) xf86CloseConsole: VT_GETMODE failed: Bad file descriptor
[   764.846] (EE) Server terminated with error (1). Closing log file.
`

alfrix, 2022-08-10:
First re-run  the installer because it's suspicious that it's trying to use fbdev instead of fbturbo
then follow:
https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/#cannot-open-virtual-console

alfrix, 2022-08-14:
OP did not reply, closing

closed 2022-08-14 by alfrix