KlipperScreen/KlipperScreen #697: [FR] KlipperScreen on machine's display!

yannvautrin, 2022-08-11:
Here is my setup:
Hypercube Evo+Raspi 4B+SKR Pro 1.1
Klipper and the gang (Moonraker, Mainsail, KlipperScreen) installed with KIAUH script, all services up and running.

At the moment, I connect with VNC to the Raspi to control the machine, update printer.cfg, etc...

I would like to add a touch screen (generic 7" HDMI touchscreen, tested and working  with the Raspi).

How can I do that and still be able to access with VNC in the same time?

(Is it possible first? or dev is required?)

Thanks!

alfrix, 2022-08-11:
> I connect with VNC to the Raspi to control the machine

most probably you are talking about SSH and not VNC, if thats the case you don´t have to anything it will keep working.

if you want to access the graphical interface remotely just search for "how to remote-desktop raspberry pi"

closed 2022-08-11 by alfrix

yannvautrin, 2022-08-12:
My issue is that, when I start the Raspi connected to the mcu and the maqchine, it starts with the Raspi OS GUI, then I have to open Chromium to access Manisail.

When I bott in console mode, then....I end hope in console mode...(CLI), not KlipperScreen...What am I doing wrong?

alfrix, 2022-08-12:
it seems that you have installed raspbian desktop, i recommend you flash Mainsail-OS
https://docs.mainsail.xyz/setup/mainsail-os
[image]

after you have mainsail os in your SD use KIAUH to install klipperscreen
https://github.com/th33xitus/kiauh

i think this is the easiest way for you.

yannvautrin, 2022-08-12:
Thanks a lot, I finally have KlipperScreen up and running!
Now, new issue... I have KlipperScreen working. Raspi is located to my local network, but  I do I find the IP adress of the Raspi to SSH it from laptop on same WLAN ?