KlipperScreen/KlipperScreen #698: [FR]

sbwilson, 2022-08-14:
**Is your feature request related to a problem? Please describe.**
In addition to showing the current Z height, would it be possible to add the current layer. Although one can calculate the layer height from the Z position, this is not possible with adaptive layer heights/ 

**Describe the solution you'd like**
Show the current and total layer count near the Z height on the job screen.

**Describe alternatives you've considered**
An alternative may be to show it in the top status bar. 

Otherwise love the project - thank you!

alfrix, 2022-08-14:
the layer info is the submenu of the job_status ([see here](https://klipperscreen.readthedocs.io/en/latest/Job_status/#position-info))
that info is not provided by moonraker and currently is calculated as you described, with the issue that you mention.
since it may not be accurate i will not place it on the main status page

the only way you will have what you want is if you do the [trick described in the docs](https://klipperscreen.readthedocs.io/en/latest/Quicktips/#layer-progress), and that method is the only one that works with the adaptive layers

closed 2022-08-14 by alfrix

vajonam, 2022-11-28:
hmm.. i see you have already added the code for it, but I dont see the Layer info in the status screen wonder why that is.

alfrix, 2022-11-28:
the current versions of moonraker/klipper/klipperscreen supports layer number but need to be setup in the slicer
Start gcode
`SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]`
layer change gcode
`SET_PRINT_STATS_INFO CURRENT_LAYER={layer_num + 1}`
if that info is not provided then layer is calculated as it was before
i will add this info to the docs

vajonam, 2022-11-28:
@alfrix  Got it thanks. I as looking for the layer in the main job_status panel, but it was in the Z sub panel.

the Z: (height in mm) is not as useful for me, maybe can we add a config option to change to Z: currt/total layer  ?  I can provide PR if you need.