KlipperScreen/KlipperScreen #699: [FR] Bring back legacy temperature setting

Drachenkaetzchen, 2022-08-14:
**Is your feature request related to a problem? Please describe.**

In KlipperScreen prior to <https://github.com/jordanruthe/KlipperScreen/pull/393> it was possible to simply set the temperature by using the big buttons "Actions", "Temperature", then "Numpad". Easy to hit, quick to navigate.

Since that change, we have to long-press the very tiny heater line and then select "Set Temperature" from a menu where it's 3 lines are as big as the whole finger. This is very cumbersome as miss-presses are the norm.

**Describe the solution you'd like**

Re-introduce the temperature panel in the "Actions" submenu.

**Describe alternatives you've considered**
Macros, but since I'm using KlipperScreen only when I'm doing maintenance or manual adjustments it would mean that I'd need different macros for each temperature.

alfrix, 2022-08-14:
> In KlipperScreen prior to https://github.com/jordanruthe/KlipperScreen/pull/393 it was possible to simply set the temperature by using the big buttons "Actions", "Temperature", then "Numpad". Easy to hit, quick to navigate.

in the current version the numpad is opened by clicking the temperature in the main menu, much faster.

the behavior of the temperature panel is being discussed at #626.

As for this FR
> Re-introduce the temperature panel in the "Actions" submenu.
Not going to happen, closing

closed 2022-08-14 by alfrix

Drachenkaetzchen, 2022-08-14:
For those who stumble over this: The solution for me was to disable additional sensors, so that the bed+extruder temperatures are big enough so that a finger can actually touch them reliably on a 5" TFT