KlipperScreen/KlipperScreen #701: [BUG]

TxBillbr, 2022-08-15:
**Describe the bug**
When using a pi as a "KlipperScreen server" with multiple printers but none on localhost, Everything works correctly except the update service.  The updater thinks it needs to update fluidd to verision ? (which is not installed on the local machine, but is up to date on the remote machine)

**To Reproduce**
Steps to reproduce the behavior:
1. Install klipper, moonraker, and KlipperScreen on a pi with attached screen. 
2. Configure KlipperScreen to have one or more non-local printers.
4. Startup KlipperScreen and navigate to System via touchscreen
5. fluidd shows running v1.17.1 -> ? and Update button is active
6. press Update
7. Screen says "Fluidd will be updated to version:?
8. Press Update
9. Screen says "Starting update for fluid...
   Error updating fluidd
  Client fluid-core/fluidd: Unable to locate update

**Expected behavior**
It would be really excellent if this worked for non-local printers. However, I suspect that the update code is using 127.0.0.1 by default and therefore confused. In a perfect world the update would indicating it is working on system <systemname> and use the moonraker API interface to take the correct steps. A reasonable work around would be to indicate that updating non-local systems is not currently supported.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/9335080/KlipperScreen.log)

BTW, Thanks for the excellent project. I really appreciate the tool. 45+ years in software development makes me appreciate good solid applications that much more.

alfrix, 2022-08-22:
>  I suspect update code is using 127.0.0.1...

not at all, it sends the command to moonraker [`"machine.update.client", {"name": program}`](https://github.com/Arksine/moonraker/blob/master/docs/web_api.md#update-client)
moonraker is the one in charge of the update, maybe checking out **moonraker.log** would reveal the issue

see here how an instance of klipperscreen on a pc that is connecting to a printer in the local network works as expected
[image]

alfrix, 2022-11-18:
i still think this is most probably something related to api and not KS, moonraker update_manager went trough a lot of updates since this, this is probably fixed, if this continues to happen and only affect KS please report back

closed 2022-11-18 by alfrix