KlipperScreen/KlipperScreen #729: [ISSUE] Disable Klipperscreen and return to xfce4 lightdm on armbian

Travis90x, 2022-09-13:
Hi everyone, after installed Klipperscreen with Kiauh, it works well, but at the reboot 
I lost possibility to use the gui on my 
Armbian_22.05.0-trunk_Tanix-tx6_bullseye_current_5.15.27_xfce_desktop
with lightdm login screen. This also happened on another armbian system.
So with putty I do:
`sudo systemctl set-default multi-user.target`

and at the next boot I can see the text mode.
Now I can:
`sudo systemctl disable KlipperScreen`
`sudo systemctl set-default graphical.target`

but at the reboot I can see only an underscore  `_`
With putty I check:
`systemctl status display-manager`
`systemctl status lightdm.service`
with the error below

I try: sudo armbian-config
and re-enable: System -> Desktop -> Enable
but return the underscore and no more GUI.

```
● lightdm.service - Light Display Manager
     Loaded: loaded (/lib/systemd/system/lightdm.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-09-14 00:41:50 CEST; 45s ago
       Docs: man:lightdm(1)
    Process: 2746 ExecStart=/usr/sbin/lightdm (code=exited, status=1/FAILURE)
   Main PID: 2746 (code=exited, status=1/FAILURE)
        CPU: 228ms

set 14 00:41:50 tanix-tx6 systemd[1]: lightdm.service: Scheduled restart job, restart counter is at 5.
set 14 00:41:50 tanix-tx6 systemd[1]: Stopped Light Display Manager.
set 14 00:41:50 tanix-tx6 systemd[1]: lightdm.service: Start request repeated too quickly.
set 14 00:41:50 tanix-tx6 systemd[1]: lightdm.service: Failed with result 'exit-code'.
set 14 00:41:50 tanix-tx6 systemd[1]: Failed to start Light Display Manager.
set 14 00:41:50 tanix-tx6 systemd[1]: lightdm.service: Triggering OnFailure= dependencies.
set 14 00:41:50 tanix-tx6 systemd[1]: lightdm.service: Failed to enqueue OnFailure= job, ignoring: Unit plymouth-quit.service not found.

```

[sudo lightdm --test-mode --debug.txt](https://github.com/jordanruthe/KlipperScreen/files/9561368/sudo.lightdm.--test-mode.--debug.txt)

after `sudo apt reinstall xfce4`
```
pi@tanix-tx6:~$ sudo startxfce4
/usr/bin/startxfce4: Starting X server
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux tanix-tx6 5.15.27-station #trunk SMP PREEMPT Wed                                                                 Mar 9 00:31:12 CET 2022 aarch64
Kernel command line: root=UUID=1319efee-48c1-47f4-9a6a-194a96bf5c12 console=ttyS                                                                0,115200 console=tty0 video=HDMI-A-1:e rootflags=data=writeback rw no_console_su                                                                spend consoleblank=0 fsck.fix=yes fsck.repair=yes net.ifnames=0
Build Date: 16 December 2021  05:08:23PM
xorg-server 2:1.20.11-1+deb11u1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Wed Sep 14 00:59:56 2022
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
modprobe: FATAL: Module g2d_23 not found in directory /lib/modules/5.15.27-stati                                                                on
/usr/lib/xorg/Xorg: symbol lookup error: /usr/lib/xorg/modules/drivers/fbturbo_d                                                                rv.so: undefined symbol: shadowUpdatePackedWeak
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error

```

Travis90x, 2022-09-14:
I solved `plymouth-quit.service not found` with 
sudo apt install plymouth plymouth-themes

and now I got

```
pi@tanix-tx6:~$ systemctl status lightdm.service
● lightdm.service - Light Display Manager
     Loaded: loaded (/lib/systemd/system/lightdm.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-09-14 21:13:55 CEST; 44s ago
       Docs: man:lightdm(1)
    Process: 15986 ExecStart=/usr/sbin/lightdm (code=exited, status=1/FAILURE)
   Main PID: 15986 (code=exited, status=1/FAILURE)
        CPU: 269ms

set 14 21:13:55 tanix-tx6 systemd[1]: lightdm.service: Scheduled restart job, restart counter is at 5.
set 14 21:13:55 tanix-tx6 systemd[1]: Stopped Light Display Manager.
set 14 21:13:55 tanix-tx6 systemd[1]: lightdm.service: Start request repeated too quickly.
set 14 21:13:55 tanix-tx6 systemd[1]: lightdm.service: Failed with result 'exit-code'.
set 14 21:13:55 tanix-tx6 systemd[1]: Failed to start Light Display Manager.
set 14 21:13:55 tanix-tx6 systemd[1]: lightdm.service: Triggering OnFailure= dependencies.
```

```
pi@tanix-tx6:~$ sudo nano /var/log/lightdm/lightdm.log
pi@tanix-tx6:~$
pi@tanix-tx6:~$ sudo rm /var/log/lightdm/lightdm.log
pi@tanix-tx6:~$ sudo nano /var/log/lightdm/lightdm.log
pi@tanix-tx6:~$ systemctl restart lightdm.service
pi@tanix-tx6:~$ sudo nano /var/log/lightdm/lightdm.log
pi@tanix-tx6:~$ systemctl status lightdm.servicesystemctl status lightdm.service^C
pi@tanix-tx6:~$ systemctl status lightdm.service
● lightdm.service - Light Display Manager
     Loaded: loaded (/lib/systemd/system/lightdm.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Wed 2022-09-14 21:13:55 CEST; 44s ago
       Docs: man:lightdm(1)
    Process: 15986 ExecStart=/usr/sbin/lightdm (code=exited, status=1/FAILURE)
   Main PID: 15986 (code=exited, status=1/FAILURE)
        CPU: 269ms

set 14 21:13:55 tanix-tx6 systemd[1]: lightdm.service: Scheduled restart job, restart counter is at 5.
set 14 21:13:55 tanix-tx6 systemd[1]: Stopped Light Display Manager.
set 14 21:13:55 tanix-tx6 systemd[1]: lightdm.service: Start request repeated too quickly.
set 14 21:13:55 tanix-tx6 systemd[1]: lightdm.service: Failed with result 'exit-code'.
set 14 21:13:55 tanix-tx6 systemd[1]: Failed to start Light Display Manager.
set 14 21:13:55 tanix-tx6 systemd[1]: lightdm.service: Triggering OnFailure= dependencies.

```

Travis90x, 2022-09-14:
I found the issue:
Klipperscreen installation with Kiauh create the file `\etc\X11\xorg.conf.d\50-fbturbo.conf`
with
```
Section "Device"
        Identifier      "Allwinner A10/A13 FBDEV"
        Driver          "fbturbo"
        Option          "fbdev" "/dev/fb0"

        Option          "SwapbuffersWait" "true"
EndSection
```

This create the errors above. 
If I move the text of that file in 
`\etc\X11\xorg.conf.d\50-fbturbo.conf\01-armbian-defaults.conf`
and delete  50-fbturbo.conf, the errore persists.
```
Section "Monitor"
	Identifier		"Monitor0"
	Option			"DPMS" "false"
EndSection
Section "ServerFlags"
	Option			"BlankTime" "0"
	Option			"StandbyTime" "0"
	Option			"SuspendTime" "0"
	Option			"OffTime" "0"
EndSection
```

If I simply delete `50-fbturbo.conf` and leave inalterate `01-armbian-defaults.conf`
all works fine: 
Klipperscreen works on Smartphone Android with XServer XSDL and XFCE works on HDMI monitor,
without `sudo systemctl set-default multi-user.target`
so remaining on ` graphical.target`

alfrix, 2022-09-17:
your issue seems to be related with the installation of `xserver-xorg-video-fbturbo` depending on your hardware you may not need it, if that is the case you should uninstall the package and probably add it to a blacklist so apt doesn't try to install it if you run the installer ever again.
you can launch klipperscreen as an application, the only catch is that KS is not resizable since it was designed to run standalone and fullscreen

alfrix, 2022-09-22:
after 1b758b85c6ecec64d2080b97afcfd04e0abe7256 you easily launch ks in a desktop environment, (still not the recommended because of the no resizable)

closed 2022-09-22 by alfrix