KlipperScreen/KlipperScreen #735: [BUG] Install Issue

Precurssor, 2022-09-27:
Trying to install and im getting a libwayland-client0 error. Im using a Pi4 with Fluiddpi. Both through manual install and Kiauh I get the same error.
[image]

Maximes21500, 2022-09-27:
I have the same problem as you :(

okruts, 2022-09-27:
I was managed to fix this issue by switching from `bullseye` to `bookworm` debian release 
Replace source lists with the following ones:
```
cat /etc/apt/sources.list
deb http://raspbian.raspberrypi.org/raspbian/ bookworm main contrib non-free rpi
#deb http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://raspbian.raspberrypi.org/raspbian/ bullseye main contrib non-free rpi

cat /etc/apt/sources.list.d/raspi.list 
deb https://deb.debian.org/debian/ bookworm main contrib non-free
# deb http://archive.raspberrypi.org/debian/ bullseye main
# Uncomment line below then 'apt-get update' to enable 'apt-get source'
#deb-src http://archive.raspberrypi.org/debian/ bullseye main
```
Then run
```
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 605C66F00D6C9793
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 0E98404D386FA1D9
sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 648ACFD622F3D138

sudo apt update
sudo apt upgrade
```
I have also issue with this package so I just reinstalled with required version
```
sudo apt-get remove libpixman-1-0
sudo apt-get install libpixman-1-0=0.40.0-1
```

Maximes21500, 2022-09-27:
Thank you for your answer.
I was able to go further but still with errors.

[image]

Maximes21500, 2022-09-27:
I found using the command: _pip install numpy_

alfrix, 2022-10-14:
workaround until the distro maintainers fix their packages:

Run `sudo apt-get install libgtk-3-0=3.24.24-4+deb11u2`

long version if the above command doesn't work or the version changes:

`apt list -a libgtk-3-0`

Will output something like this
```
libgtk-3-0/stable 3.24.24-4+rpt7+deb11u2 arm64
libgtk-3-0/stable 3.24.24-4+deb11u2 arm64
libgtk-3-0/stable 3.24.24-4+rpt7+deb11u2 armhf
libgtk-3-0/stable 3.24.24-4+deb11u2 armhf
```
The versions with  "rpt7" are the problem, select the other version

[source](https://forums.raspberrypi.com/viewtopic.php?p=2045089)

closed 2022-10-14 by alfrix