KlipperScreen/KlipperScreen #736: [BUG] network doesn't show any networks. Workaround in troubleshooting does not work

BimDub, 2022-09-27:
**Describe the bug**
When using a non-raspberry pi system running Debian, the Network page in klipperscreen just shows the currently connected network. It does not allow you to disconnect, scan for other networks, or do anything other than show your host name and current IP address

**To Reproduce**
Steps to reproduce the behavior:
1. Install Debian on a normal PC
2. Run KlipperScreen and try to change network settings
3. See error

**Expected behavior**
Being able to scan for networks or disconnect from current network

**Screenshots**
https://i.ibb.co/svdvZJF/Screenshot-20220927-085101.png

**Logfile**
--------------------KlipperScreen Log Start--------------------
Git Version: v0.2.6-23-g35571fc-dirty
2022-09-27 08:35:51,806 [screen.py:main()] - KlipperScreen version: v0.2.6-23-g35571fc-dirty
2022-09-27 08:35:51,810 [config.py:get_config_file_location()] - Passed config file: /home/pi/KlipperScreen.conf
2022-09-27 08:35:51,812 [config.py:get_config_file_location()] - Found configuration file at: /home/pi/KlipperScreen/ks_includes/defaults.conf
2022-09-27 08:35:51,813 [config.py:__init__()] - Config path location: /home/pi/KlipperScreen/ks_includes/defaults.conf
2022-09-27 08:35:51,903 [config.py:__init__()] - Configuration validated succesfuly
2022-09-27 08:35:51,904 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2022-09-27 08:35:51,904 [config.py:__init__()] - Detected language: None
2022-09-27 08:35:51,911 [screen.py:__init__()] - OS Language: en_US.UTF-8
2022-09-27 08:35:52,280 [screen.py:__init__()] - Screen resolution: 960x540
2022-09-27 08:35:52,281 [KlippyGtk.py:__init__()] - img width: 60 height: 58
2022-09-27 08:35:53,460 [screen.py:_load_panel()] - Loading panel: splash_screen
2022-09-27 08:35:53,460 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/splash_screen.py
2022-09-27 08:35:53,611 [screen.py:show_panel()] - Attaching panel splash_screen
2022-09-27 08:35:53,616 [screen.py:show_panel()] - Showing back. count: 0
2022-09-27 08:35:53,627 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2022-09-27 08:35:53,656 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2022-09-27 08:35:53,668 [screen.py:set_screenblanking_timeout()] - DPMS State FAIL
2022-09-27 08:35:53,669 [screen.py:set_screenblanking_timeout()] - Not using DPMS
2022-09-27 08:35:54,377 [screen.py:initial_connection()] - Default printer: None
2022-09-27 08:35:54,377 [screen.py:connect_printer()] - Connecting to printer: Printer
2022-09-27 08:35:54,378 [screen.py:show_panel()] - Attaching panel splash_screen
2022-09-27 08:35:54,388 [screen.py:show_panel()] - Showing back. count: 0
2022-09-27 08:35:54,389 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2022-09-27 08:35:54,394 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 08:35:54,419 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2022-09-27 08:35:54,434 [KlippyWebsocket.py:connect()] - Starting websocket thread
2022-09-27 08:35:54,436 [screen.py:connect_printer()] - Connected to printer: Printer
2022-09-27 08:35:54,451 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2022-09-27 08:35:54,451 [KlippyWebsocket.py:on_open()] - Self.connected = False
2022-09-27 08:35:54,452 [_logging.py:warning()] - websocket connected
2022-09-27 08:35:54,505 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 08:35:54,526 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 08:35:54,561 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 08:35:54,576 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
2022-09-27 08:35:57,373 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 08:35:57,400 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 08:35:57,428 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 08:35:57,455 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
2022-09-27 08:36:00,372 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 08:36:00,393 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 08:36:00,416 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 08:36:00,443 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
2022-09-27 08:36:03,373 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 08:36:03,388 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
[... 249 lines cut ...]
2022-09-27 08:50:57,747 [screen.py:show_panel()] - Attaching panel splash_screen_
2022-09-27 08:50:57,750 [screen.py:show_panel()] - Showing back. count: 1
2022-09-27 08:50:57,754 [menu.py:arrangeMenuItems()] - Evaluating item: power
2022-09-27 08:50:57,754 [menu.py:evaluate_enable()] - Template: '{{ printer.power_devices.count > 0 }}'
2022-09-27 08:50:57,757 [menu.py:arrangeMenuItems()] - Evaluating item: network
2022-09-27 08:50:57,759 [menu.py:arrangeMenuItems()] - Evaluating item: system
2022-09-27 08:50:57,759 [menu.py:evaluate_enable()] - moonraker is_connected True
2022-09-27 08:50:57,761 [menu.py:arrangeMenuItems()] - Evaluating item: settings
2022-09-27 08:50:57,763 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen', 'splash_screen_']
2022-09-27 08:50:58,759 [screen_panel.py:menu_item_clicked()] - ### Creating panel network : network {'name': "{{ gettext('Network') }}", 'icon': 'network', 'panel': 'network', 'method': False, 'confirm': False, 'enable': True, 'params': {}}
2022-09-27 08:50:58,760 [screen.py:_load_panel()] - Loading panel: network
2022-09-27 08:50:58,760 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/network.py
2022-09-27 08:50:58,823 [screen.py:show_panel()] - Attaching panel splash_screen__network
2022-09-27 08:50:58,829 [screen.py:show_panel()] - Showing back. count: 2
2022-09-27 08:50:58,831 [screen.py:show_panel()] - Error attaching panel:
'NoneType' object has no attribute 'rescan'
Traceback (most recent call last):
  File "/home/pi/KlipperScreen/screen.py", line 350, in show_panel
    self.panels[panel_name].activate()
  File "/home/pi/KlipperScreen/panels/network.py", line 455, in activate
    self.reload_networks()
  File "/home/pi/KlipperScreen/panels/network.py", line 449, in reload_networks
    self.wifi.rescan()
AttributeError: 'NoneType' object has no attribute 'rescan'
2022-09-27 08:50:58,833 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen', 'splash_screen_', 'splash_screen__network']

alfrix, 2022-09-27:
this will fix the error in the logs, if the issue continues after the update please add another logfile here or in another issue

closed 2022-09-27 by alfrix

BimDub, 2022-09-27:
Still doing the exact same thing on the screen. I also tried changing from NetworkManager to wpa_supplicant as my network configuration tool, same thing no matter which I use. I'd prefer to use NetworkManager if possible, though.

--------------------KlipperScreen Log Start--------------------
Git Version: v0.2.6-24-g79b3429
2022-09-27 18:22:59,109 [screen.py:main()] - KlipperScreen version: v0.2.6-24-g79b3429
2022-09-27 18:22:59,111 [config.py:get_config_file_location()] - Passed config file: /home/pi/KlipperScreen.conf
2022-09-27 18:22:59,111 [config.py:get_config_file_location()] - Found configuration file at: /home/pi/KlipperScreen/ks_includes/defaults.conf
2022-09-27 18:22:59,112 [config.py:__init__()] - Config path location: /home/pi/KlipperScreen/ks_includes/defaults.conf
2022-09-27 18:22:59,188 [config.py:__init__()] - Configuration validated succesfuly
2022-09-27 18:22:59,188 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2022-09-27 18:22:59,189 [config.py:__init__()] - Detected language: None
2022-09-27 18:22:59,202 [screen.py:__init__()] - OS Language: en_US.UTF-8
2022-09-27 18:22:59,323 [screen.py:__init__()] - Screen resolution: 960x540
2022-09-27 18:22:59,325 [KlippyGtk.py:__init__()] - img width: 60 height: 58
2022-09-27 18:22:59,814 [screen.py:_load_panel()] - Loading panel: splash_screen
2022-09-27 18:22:59,815 [screen.py:_load_panel()] - Panel path: /home/pi/KlipperScreen/panels/splash_screen.py
2022-09-27 18:22:59,965 [screen.py:show_panel()] - Attaching panel splash_screen
2022-09-27 18:22:59,971 [screen.py:show_panel()] - Showing back. count: 0
2022-09-27 18:22:59,982 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2022-09-27 18:23:00,009 [screen.py:set_screenblanking_timeout()] - Changing screen blanking to: 3600
2022-09-27 18:23:00,018 [screen.py:set_screenblanking_timeout()] - DPMS State FAIL
2022-09-27 18:23:00,019 [screen.py:set_screenblanking_timeout()] - Not using DPMS
2022-09-27 18:23:00,651 [screen.py:initial_connection()] - Default printer: None
2022-09-27 18:23:00,652 [screen.py:connect_printer()] - Connecting to printer: Printer
2022-09-27 18:23:00,653 [screen.py:show_panel()] - Attaching panel splash_screen
2022-09-27 18:23:00,660 [screen.py:show_panel()] - Showing back. count: 0
2022-09-27 18:23:00,661 [screen.py:show_panel()] - Current panel hierarchy: ['splash_screen']
2022-09-27 18:23:00,666 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 18:23:00,697 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/access/oneshot_token
2022-09-27 18:23:00,715 [KlippyWebsocket.py:connect()] - Starting websocket thread
2022-09-27 18:23:00,719 [screen.py:connect_printer()] - Connected to printer: Printer
2022-09-27 18:23:00,726 [KlippyWebsocket.py:on_open()] - Moonraker Websocket Open
2022-09-27 18:23:00,727 [KlippyWebsocket.py:on_open()] - Self.connected = False
2022-09-27 18:23:00,727 [_logging.py:warning()] - websocket connected
2022-09-27 18:23:00,796 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 18:23:00,821 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 18:23:00,860 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 18:23:00,883 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
2022-09-27 18:23:04,170 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 18:23:04,201 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 18:23:04,229 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 18:23:04,256 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
2022-09-27 18:23:07,170 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 18:23:07,199 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 18:23:07,226 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 18:23:07,254 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
2022-09-27 18:23:10,169 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 18:23:10,186 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 18:23:10,201 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 18:23:10,217 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
2022-09-27 18:23:13,170 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 18:23:13,197 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 18:23:13,225 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 18:23:13,255 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
[... cut ...]
2022-09-27 18:23:55,740 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker
2022-09-27 18:23:58,657 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/server/info
2022-09-27 18:23:58,700 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/machine/device_power/devices
2022-09-27 18:23:58,730 [KlippyRest.py:send_request()] - Sending request to http://127.0.0.1:7125/printer/info
2022-09-27 18:23:58,754 [screen.py:_update_splash_screen()] - Unable to get printer info from moonraker

commit 79b34298 (2022-09-27): network: fix panel not loading if the wifi module has not ben initialized

close #736