KlipperScreen/KlipperScreen #744: [FR] Docker container?

kerta1n, 2022-10-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Will there be a Docker container for Klipperscreen?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
This way users who do run their 3D printing SW in Docker can experiment with different setups without installing all of the dependencies onto the host system like all of the X server dependencies. Also, this container could have a switch between displaying to its hosts' physical display, and only running a display server for clients (like Android devices) to connect to.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
mkuf/prind has a container, but it involves using his compose stack.

**Additional context**
Add any other context or screenshots about the feature request here.

alfrix, 2022-11-02:
Use [prind](https://github.com/mkuf/prind)

closed 2022-11-02 by alfrix

kerta1n, 2022-11-02:
As I listed earlier, I had tried to use prind, but it was not that easy to use since I had to use Klipper/Moonraker/Mainsail from his stack, and was not as configurable as I'd hoped it to be. Is that the only solution there is then?