KlipperScreen/KlipperScreen #748: [BUG] Bl-touch does not raise z axis to deploy pin

kyledavis417, 2022-10-05:
**Bltouch can't deploy when z calibrate has been aborted and ran again**
If you run a z calibrate and abort and try to run the z calibrate again the probe can be too close to the bed and the pin can't deploy all the way causing you to have to home to the printer to try again.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to z calibrate 
2. Click on start
4. move the nozzle down
5. Abort the process
6. go to z calibrate
7. click on start
8. pin hits the bed 
9. error in console that pin failed to deploy

**Expected behavior**
when you click start in z calibrate, if you have a bl touch the printer should rehome z axis before allowing the offset to be adjusted if the printer is homed.

alfrix, 2022-10-08:
this behavior is part of klipper, i don't think it should be addressed on the UI but in klipper itself

tinymachines3d, 2022-10-12:
@alfrix that's not the issue. Klipper Screen zcalibrate.py does not have a check for printhead location before starting z calibration. It only checks to see if the printer has been homed. If homed, it runs the script from wherever the printhead is. Say, for example, the user auto homes the machine, then homes only the Y axis and puts the printhead off the build plate (CORE XY), when you start Z calibrate, nozzle will crash.

alfrix, 2022-10-21:
zcalibrate does move to a configured or calculated position when doing probe_calibrate the other methods shouldn't require reposition
https://github.com/jordanruthe/KlipperScreen/blob/master/panels/zcalibrate.py#L148

tinymachines3d, 2022-11-04:
@alfrix 

If user aborts from Z Calibrate, uses actions>move, Z does not raise before moving to position. So, if the print head is close to the bed (after adjusting Z offset down, then clicking abort) bl touch fails to deploy because it hits the bed before any Z movement. 

Video link and klipperscreen log attached. 

https://youtu.be/2WYz3t5oMEM

[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/9940991/KlipperScreen.log)

closed 2022-11-07 by alfrix

tinymachines3d, 2022-11-08:
@alfrix This only works with machines not using homing_override. For example if a machine needs to home Y before X, homing_override is required and safe_z_home has to be turned off. In this case a force Z raise would need to be ran before Z calibrate is ran again. The issue is the same as before [23b64b1](url) was pushed.

commit 23b64b16 (2022-11-07): zcalibrate: use safe_z z_hop if defined close #748