KlipperScreen/KlipperScreen #749: When using the KlipperScreen I lose Wifi signal to my Pi [BUG]

Bakes70, 2022-10-06:
I have installed a BTT Pi TFT50 screen on my Raspberry Pi and it works as a Klipperscreen to my 3D printer that previously ran Klipper + Mainsail without a screen display.

The problem I have is that with the screen turned on I can’t connect to my printer via wifi + Mainsail. As soon as I disconnect the screen I can connect to the printer again via wifi + Mainsail.

I have tried removing the Pi from the back of the screen. I have dimmed the display (so it draws less power). And I have even covered the screen with aluminium foil (in case there is electrical interference blocking the wifi signal). None of these tests have allowed me to connect to the printer via wifi with the screen connected.

It seems this problem has happened for other people to. Can you share any tips that may fix the problem as I do like the idea of using a KlipperScreen and the BTT Pi TFT50 is listed as suitable hardware?

Thanks for any assistance you can provide.

P.S. I have tried to access the klipperscreen.log file but had difficulty locating it.

Andrew

alfrix, 2022-10-08:
some people have no issues and others are affected by it, it doesn't help that there are various revisions of the screen and some models of the Pi behave differently, it appears to be a hardware issue not fixable by software

Closing since this is not a KlipperScreen Bug

closed 2022-10-08 by alfrix

Bakes70, 2022-10-11:
Thanks anyway. It is helpful to know that this issue is more likely a
hardware issue.

 

I will contact Big Tree Tech to see if they have any suggestions.

 

Regards Andrew 

Wifi signal to my Pi [BUG] (Issue #749)

 

<https://github.com/jordanruthe/KlipperScreen/issues/749#issuecomment-127232
F3SZANCNFSM6AAAAAAQ6IZUG4> .
<https://github.com/notifications/beacon/A3OODJNYCGFKCXXHHYNTSPTWCF3SZA5CNFS
M6AAAAAAQ6IZUG6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTS