KlipperScreen/KlipperScreen #750: [FR] Adding of Dockable_probe option for Z-Calibrate

Geeps-SA, 2022-10-06:
**Is your feature request related to a problem? Please describe.**
When using Mental's Dockable_probe.py module which you use a Dockable_probe section in place of a normal Probe in the printer.cfg, the Z-Calibrate naturally does not work.

**Describe the solution you'd like**
It would be great if you could add dockable_probe as another option to look for in the printer.cfg 

**Describe alternatives you've considered**
I made a copy of the zcalibrate.py panel and replaced all mentions to "probe" with "dockable_probe" and replaced the original. Which does work.

**Additional context**
[image]

alfrix, 2022-11-02:
at this moment i don't think modules outside mainline klipper will be supported
however i will offer you a solution that can update the rest of KS:
copy the panel as `zcalibrate_custom.py`, do your changes and replace the default by creating a menu in KlipperScreen.conf
```
[menu __main config zoffset]
name: {{ gettext('Z Calibrate') }}
icon: z-farther
panel: zcalibrate_custom
```

Geeps-SA, 2022-11-04:
Morning Jordan

Thank you for the work around which seems to work but sadly my replacing
the "probe" mentions with "dockable_probe" does not work with the latest
update not sure if you could help me out on how to get it working again?
Anyways, have a good day.

Kind regards
Grant

alfrix, 2022-11-04:
i'm not Jordan, try this:
[zcalibrate.zip](https://github.com/jordanruthe/KlipperScreen/files/9937468/zcalibrate.zip)

Geeps-SA, 2022-11-04:
Hi Alfredo,
Sorry about that....Firstly thank you for your help so far. I've tried your file it does get one step further ...as in I can see it says "Probe offset" under the start button but if I press the start nothing happens further. Not sure if there is a log file or something I can send you if it'll help, if you have any other ideas or time to work this out.
[image]

alfrix, 2022-11-09:
i'm sorry for the delay, could you please try this: [zcalibrate.zip](https://github.com/jordanruthe/KlipperScreen/files/9971531/zcalibrate.zip)

it should have what you want plus some auto z_hop if probe is too close after an abort

Geeps-SA, 2022-11-09:
No worries about the delay, your new file seems to work, couple of minor things but I'd say that on my side:
a) Not sure where I could add the x_position and y_position as I think it's looking for them in the safe_z_home which I don't use. So it takes the x & y max and divides it in half which in my case is not the center of the bed but I worked around it by minusing the difference.
b) I think the z_hop you refer to is also only active if you have safe_z_home in your config.

So again from my side thank you so much for all the help !!

alfrix, 2022-11-09:
it will try to input some sane values even without safe_z_home, the idea is to probe the bed and not crash outside, you can configure the xy position in klipperscreen.conf if you prefer a particular probing position

Geeps-SA, 2022-11-10:
Morning Alfredo, thank you again for this, I'll look into adding those
positions into K/Screen config. Hopefully your zcalibrate file gets
accepted and then I can do away with the zcalibrate_custom panel.

alfrix, 2022-11-10:
it's already on master

closed 2022-11-10 by alfrix