KlipperScreen/KlipperScreen #752: Stuck on "Connecting to Printer" on Fresh install of everything, with work-a-round/fix.

skellatore, 2022-10-10:
Hi Dev's

Fresh install and getting nowhere. 
Everything generally works but the KlipperScreen basically does nothing and sits there saying "Connecting to Printer". I suspect it's to do with this error:
2022-10-10 11:12:03,631 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'is_paused'
Traceback: ['  File "/home/pi/KlipperScreen/screen.py", line 1024, in init_printer\n    self.printer.process_update(data[\'result\'][\'status\'])\n', '  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 138, in process_update\n    self.evaluate_state()\n', '  File "/home/pi/KlipperScreen/ks_includes/printer.py", line 150, in evaluate_state\n    if self.data[\'pause_resume\'][\'is_paused\']:\n']
NoneType: None

I realised it's because the key pause_resume didn't exist which meant is_paused wouldnt exist and so changing out line 150 in ks_includes/printer.py from:
`if self.data['pause_resume']['is_paused']:`
to:
`if self.data.get('pause_resume').get('is_paused'):`
avoided evaluating a NoneType and moved into the elif statement below. This allowed everything to work correctly.

Log for good measure
[KlipperScreen.log](https://github.com/jordanruthe/KlipperScreen/files/9745122/KlipperScreen.log)

alfrix, 2022-10-10:
this is caused by your klipper config is missing `[pause_resume]`
this is a bug, although the module is pretty useful and recommended, it shouldn't prevent KS from starting

skellatore, 2022-10-10:
Oh for sure I will be adding pause/resume... but this is a fresh build and wanted to test the minimum klipper config was valid before adding in all the pretties :)

Using the get method, whilst being slower to eval, will allow the if statement to evaluate if the key doesn't exist. Easy fix :)

commit 978bb72e (2022-10-16): Fix: Printer not connecting if [pause_resume] is missing

fix #752
closed 2022-10-16 by alfrix