KlipperScreen/KlipperScreen #753: Touchscreen not working on only parts of 2 screens

the1snm, 2022-10-12:
I've been using KS on my other machine for multiple months with a BTT TFT50 with no issues. I recently purchased a BTT TFT43 to use on another machine I converted to klipper. Everything works fine with it except on 2 screens, the touch is not recognized on the very bottom. When  in the Fine Tune & Printer Move screens the distance buttons on the bottom do not respond to any input (see images for red box). On the Fine Tune screen I can touch the tops of the buttons (blue box in image) and it will register the button above it i.e. the top of the 1%/5% will decrease speed by 1%. I can manually click the +/- for all 6 buttons in the middle and the reset at the tops fine. On other screens recognize touches on the very bottom fine like the settings etc. I've tested this on a new fresh install on a different RPi4 with the same results. With the TFT50 which has the same resolution there is no issue with these parts of the screen.

[image]
[image]

alfrix, 2022-10-14:
this is a hw issue, you may want to try to calibrate the touchscreen as i explained in #517. is not easy but you can use the [transformation matrix](https://klipperscreen.readthedocs.io/en/latest/Hardware/#touchscreen-touch-rotation) to do it.
 
 [more info in the ubuntu doc about libinput matrixes](https://wiki.ubuntu.com/X/InputCoordinateTransformation)

Closing since this is not fixable by changing KlipperScreen code

closed 2022-10-14 by alfrix

the1snm, 2022-10-18:
I probably should have waiting a little longer before posting this question as not long after doing so I ordered a replacement from Amazon. It arrived this past weekend and I plugged it in. Worked perfectly right out of the box. Thanks for taking the time to reply, sorry for using up a few minutes of your time on this but maybe this will help someone else with a similar issue searching for answers.