KlipperScreen/KlipperScreen #755: [BUG] KlipperScreen on a Pi4 with a Hyperpixel 4.0 screen only in portrait mode

joris57, 2022-10-14:
I am trying to run KlipperScreen on a Pi4 with a Hyperpixel 4.0 screen.

After adding the following lines to /boot/config.txt, the Pi4 displays in landscape mode, both in console and desktop mode with the touch screen functioning accordingly.

```
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
```

KlipperScreen, however, is still displaying in portrait mode (with touch screen in landscape mode).

All software is at it's most recent release.

alfrix, 2022-10-14:
Duplicate of #665

closed 2022-10-14 by alfrix

joris57, 2022-10-14:
For future reference :

To change the rotation of KlipperScreen on a Pi 4B with HyperPixel 4 from portrait to landscape:

Add to /boot/config.txt

```
dtoverlay=vc4-kms-dpi-hyperpixel4
dtparam=rotate=90,touchscreen-swapped-x-y,touchscreen-inverted-y
```

Add to /usr/share/X11/xorg.conf.d/88-hyperpixel4.conf

```
Section "Monitor"
    Identifier "DPI-1"
    Option "Rotate" "Right"
EndSection
```