KlipperScreen/KlipperScreen #760: [BUG] KlipperScreen.conf patch stop working

Guilouz, 2022-10-20:
Since recent Moonraker path changes the path ~/klipper_config/KlipperScreen.conf have stopped to work.

alfrix, 2022-10-21:
it's not a bug, KS should be able to read and write to the old folder no problem, and the user can use other config locations with -c or even continue using that location with a bunch of symlinks (see data-path-fix script from moonraker)

i will change this to an FR for adding the new config path to the search, for new machines/installs

ayesim, 2022-10-23:
if system working properly, after update moonraker creates links under ~/printer_data/  . New install moonraker creates folders. either cases KlipperScreen uses ~/klipperscreen.conf and both cases are working.

Guilouz, 2022-10-23:
Yes I know that. This FR is just for klipperscreen search also in printer_data/config folder to have access to klipperscreen.conf on mainsail/fluidd web interface.

For now I have fixed this by changing klipperscreen code.

ayesim, 2022-10-23:
if they implement such as ~/printer_1_data/config/, ~/printer_2_data/config/ multi printer instances. it would be great benefit for KlipperScreen, because we'd have different theme for each specific printer.

alfrix, 2022-10-28:
fixed with c5413e51c9f0175c0d2420ce8d7cb128c004288d

closed 2022-10-28 by alfrix