KlipperScreen/KlipperScreen #761: [FR] Popup Notification Command

Nitro2k5, 2022-10-20:
Hi, i would like to request a new feature. 

It would be nice to send a popup notification to KlipperScreen via command in a gcode macro.

Preferably via standalone command:
KLIPPERSCREEN_NOTIFICATION TEXT="Hello World!" (variables possible)
(optional append CLOSE_SELF=True SHOW_TIME=10)
(optional append TEXT_COLOR=blue BACK_COLOR=white)
or some variation of it. 

Or if this is not possible, use the normal M117 function instead which contains a special syntax (buzzword). 

M117 *[KLIPPERSCREEN]*Hello World!
Output: Hello World!

And after KlipperScreen has recognized the buzzword, it displays the notification banner and clears the M117 message itself with an empty M117 command. 

This feature would be great and would make working without webinterface much easier for custom macro solutions. 

BR

Guilouz, 2022-10-20:
It's already possible with [respond] function in printer.cfg.

And use RESPOND MSG="Your text" in your macros.

Nitro2k5, 2022-10-21:
Thanks a lot for your help!
You might want to point this out on the QuickTips documentation.
I don't think this is documented anywhere.

closed 2022-10-21 by Nitro2k5