KlipperScreen/KlipperScreen #764: [BUG] Fine Tuning panel KlipperScreen.conf z_babystep_values not always respected

northcoastam, 2022-10-24:
**Describe the bug**
Tuning panel KlipperScreen.conf z_babystep_values not respected

I think in fine_tune.py the value of self.bs_delta needs to be reset after loading custom z_babystep_values from KlipperScreen.conf and setting value of `self.bs_deltas = bs` on line 34

**To Reproduce**
Steps to reproduce the behavior:
1. Add z_babystep_values item to KlipperScreen.conf.  **Ensure second value is different to default "0.05**"
      Example:
      
```
      [printer Ender 3 v2]
      
      # Define the z_babystep intervals in a CSV list. Currently only 2 are supported
      z_babystep_values: 0.01, 0.04 
```

3. Restart KlipperScreen, begin a print and go to Fine Tuning panel
4. See error

- Neither the 0.01 nor the 0.04 label is highlighted as active.  Tapping the buttons does not make them highlighted.
- Tap the Z+ or Z- button and notice the z-offset changes by 0.05.

See attached screenshot below

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Logfile**
Attach your [klipperscreen.log](https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/)

excerpt from log file:
2022-10-25 07:57:17,016 [fine_tune.py:change_bs_delta()] - ### BabyStepping 0.04
2022-10-25 07:57:17,017 [functions.py:logging_exception_handler()] - Uncaught exception <class 'KeyError'>: 'bdelta0.05'
Traceback: ['  File "/home/martin/KlipperScreen/panels/fine_tune.py", line 168, in change_bs_delta\n    self.labels[f"bdelta{self.bs_delta}"].get_style_context().remove_class("distbutton_active")\n']
NoneType: None

northcoastam, 2022-10-24:
I just made a PR for this - probably a bit over the top for one line of code - I've never made a PR before, apologies for the noise

alfrix, 2022-10-28:
> apologies for the noise

you did everything right, issue with description and PR to fix it, Thank you!

closed 2022-10-28 by alfrix

northcoastam, 2022-10-28:
Thank you.  I can’t believe how good KlipperScreen is.  I recently changed to Klipper, running headless on an old NUC.  When I learned you can run KlipperScreen an old android phone over wifi, I though it would be difficult to set up and maybe a bit flakey.  I works great and was so easy thanks to the excellent documentation.  A much better interface than Mainsail for a small touchscreen, it’s a pleasure to use.