KlipperScreen/KlipperScreen #767: [BUG]  v0.2.8 can't boot

yhira1023, 2022-10-28:
After update v0.2.8 ,KlipperScreen can't boot
The touch pannel shows "[FAILD] Failed to start starts mjpg-streamer on startup."

**Logfile**
--------------------KlipperScreen Log Start--------------------
Git Version: v0.2.8-0-gf6e4e23-dirty
2022-10-28 15:43:39,754 [screen.py:main()] - KlipperScreen version: v0.2.8-0-gf6e4e23-dirty
2022-10-28 15:43:39,756 [config.py:__init__()] - Config path location: /home/boyon/KlipperScreen/ks_includes/defaults.conf
2022-10-28 15:43:39,825 [config.py:__init__()] - Configuration validated succesfuly
2022-10-28 15:43:39,825 [config.py:__init__()] - Configured printers: [
  {
    "Printer": {
      "moonraker_host": "127.0.0.1",
      "moonraker_port": "7125",
      "moonraker_api_key": ""
    }
  }
]
2022-10-28 15:43:39,827 [config.py:create_translations()] - Available lang list ['da', 'de_DE', 'en', 'es', 'fr_FR', 'he_IL', 'hu_HU', 'it', 'ko_KR', 'pl_PL', 'ru_RU', 'sv_SE'
, 'tr_TR', 'uk_UA', 'zh_CN', 'zh_TW']
2022-10-28 15:43:39,869 [config.py:create_translations()] - Selected lang: None OS lang: ja_JP.UTF-8
2022-10-28 15:43:39,869 [config.py:install_language()] - lang: None not found

alfrix, 2022-10-28:
fixed with f749969c07594b84a5b05cc37d49078364d609db

closed 2022-10-28 by alfrix

yhira1023, 2022-10-28:
Waw! very Quicky!
Thanks.