KlipperScreen/KlipperScreen #770: [BUG] Keyboard Symbol ⌫ not recognized

Guilouz, 2022-10-29:
On RaspiOS Lite ⌫ keyboard symbol is not recognized :

[image]

On Ubuntu LTS no problem :

[image]

Any idea what is missing ?

alfrix, 2022-10-30:
not really a "bug", you are missing a font glyph, installing a font package like fonts-freefont-ttf should fix this
don't worry too much, i have plans to remove them and replace them with icons like the keypad
[image]

Guilouz, 2022-10-30:
Thanks for your reply.

closed 2022-10-30 by Guilouz