KlipperScreen/KlipperScreen #776: [BUG] KlipperScreen-install.sh - Can not install - Building wheel for numpy (PEP 517) ... error

cottec, 2022-11-04:
Unfortunately the script does not run thorugh and is stuck during
Building wheels for collected packages: numpy
Building wheel for numpy (PEP 517) 
See attached log...

Unfortunately i can't use my new V400 without it, maybe someone has an idea how to fix that?

[numpy.txt](https://github.com/jordanruthe/KlipperScreen/files/9933850/numpy.txt)

Cheers

alfrix, 2022-11-04:
try
```
cd ~/.KlipperScreen-env
source bin/activate
pip install --upgrade pip setuptools wheel
```
reinstall

and by the way, you are installing https://github.com/Guilouz/KlipperScreen-Flsun-Speeder-Pad so you should report issues to that repo

closed 2022-11-04 by alfrix

cottec, 2022-11-04:
thanks mate,
he doesnt have any clue what's wrong.
and looks like it's only changed config files in his variant for a specific printer, nothing else.
so i thought it might be a general issue.

alfrix, 2022-11-04:
still failing different error...
i can think of 2 possible solutions
option 1
`sudo apt install python-numpy`
reinstall KS
option 2
```
cd ~/.KlipperScreen-env
source bin/activate
pip install numpy
```
that should install the lastest version of numpy (that doesn't support python 3.7 and cannot be installed by default)
if that worked follow with
```
pip install matplotlib
```

remove both matplotlib and numpy from KlipperScreen/scripts/KlipperScreen-requirements.txt to prevent them from being downgraded

cottec, 2022-11-04:
first one ended in lost connection after like 30minutes
second one failed after ~10 minutes
[numpy3.txt](https://github.com/jordanruthe/KlipperScreen/files/9937338/numpy3.txt)

cottec, 2022-11-04:
wow so i literally tried the whole day playing around with this but nothing worked.

do yu have any other idea mate? :(

alfrix, 2022-11-04:
i would check free space on the device, building packages needs space
then 
```
cd ~/.KlipperScreen-env
source bin/activate
pip install --no-use-pep517 numpy==1.21.4
```
> fresh installation

it's not a fresh install unless you removed the enviroment (deleted .KlipperScreen-env)

this is not a bug in KlipperScreen this is an issue with a dependency in your system, most people would end up with a full reinstall

cottec, 2022-11-04:
Checked space, seems good,
[image]

Again fails, on page 42 you can see the separate no-use pep517
[MobaXterm_192.168.178.31pi_20221104_234300.docx](https://github.com/jordanruthe/KlipperScreen/files/9942386/MobaXterm_192.168.178.31pi_20221104_234300.docx)

I dont know what to do anymore, it's nothing special and several people successfully flashed their FLSUN SpeederPad with Guilouz instructions...
https://github.com/Guilouz/Klipper-Flsun-Speeder-Pad

this is what a did in detail, i can not find any difference to what is proposed there...

`pi:flsun

Prepare
Update Ubuntu dependencies
sudo apt update
sudo apt full-upgrade
y
sudo reboot

Change Timezone
sudo timedatectl set-timezone Europe/Berlin

Change Wi-Fi Location
sudo iw reg set DE
sudo sed -i 's/^REG.*=$/&DE/' /etc/default/crda
sudo sed -i '/^exit 0/ d' /etc/rc.local
echo -e "iw reg set DE\nexit 0" | sudo tee -a /etc/rc.local
sudo reboot

Delete Flsun Builds
git clone https://github.com/th33xitus/kiauh.git
./kiauh/kiauh.sh

Remove all
3 remove
5 klipperscreen
3 mainsail
2 moonraker
1 klipper
B exit
Q exit

Delete old folders
sudo rm -rf /home/pi/.moonraker_database_1
sudo rm -rf /home/pi/.moonraker_database_2
sudo rm -rf /home/pi/.moonraker_database_3
sudo rm -rf /home/pi/klipper_config
sudo rm -rf /home/pi/klipper_logs
sudo rm /home/pi/.gitconfig
sudo rm /home/pi/savedVariables1.cfg
sudo rm /home/pi/savedVariables2.cfg
sudo rm /home/pi/savedVariables3.cfg

Install Official Builds (1 instance)
./kiauh/kiauh.sh
1 install
1 klipper
1 python 2.7
y
2 moonraker
3 mainsail
n
(5) klipperscreen
B exit
Q exit

I left out this one at this try of course and installed dirctly from kiauh the official project
# Custom Guilouz KlipperScreen
# git clone https://github.com/Guilouz/KlipperScreen-Flsun-Speeder-Pad.git
# sudo mv /home/pi/KlipperScreen-Flsun-Speeder-Pad /home/pi/KlipperScreen
# cd ~/KlipperScreen
# ./scripts/KlipperScreen-install.sh

Mount USB
cd ~
ln -s /home/pi/gcode_files/USB-Disk /home/pi/printer_data/gcodes/USB-Disk
sudo reboot`

cottec, 2022-11-05:
@alfrix anything i forgot / did wrong?

alfrix, 2022-11-05:
try editing ~/KlipperScreen/scripts/KlipperScreen-requirements.txt and changing the numpy version to 1.20.3 or even 1.19.5, i think matplotlib 3.5.0 needed 1.17 so you can go back a little

this is deeply frustrating numpy has been a constant source of issues since last year, the only real benefit from using it alog with matplotlib is the 3d graph, which i'm considering nuking it and making it a built-in 2d map, for my sanity

cottec, 2022-11-05:
no success with neither 1.19.5 nor 1.17.0

cottec, 2022-11-12:
@alfrix by any chance, could you reopen this issue so I can see when you fix the issue?

alfrix, 2022-11-12:
numpy and matplotlib are not required for this repo anymore, so this issue is no longer valid
the commits in question are 88461910c99248e29d107bfe24561513a3528fc1, ce6158ad91ca1f556a20c3b429cc7ed241c69931, f5b8ae1002aeb01e7e4810f8ada6fad6b96b9d59

cottec, 2022-11-13:
@alfrix
sweeeet, can confirm this is working!!!!!
Thank you very very much