KlipperScreen/KlipperScreen #779: Unable to Rotate Screen with RatOS

ajquick, 2022-11-07:
I am using a default install of RatOS, which includes KlipperScreen.

I am attempting to rotate the screen (it is currently 180 degrees out). 

Since I am using RaspberryPi in an essentially headless configuration, I'm not sure how I am to detect the display and thus force the rotation. I have SSH'ed into the Pi and run the following command:

````DISPLAY=:0.0 xrandr --verbose````

This outputs:

````
xrandr: Failed to get size of gamma for output default
Screen 0: minimum 1024 x 600, current 1024 x 600, maximum 1024 x 600
default connected 1024x600+0+0 (0x2e1) normal (normal) 0mm x 0mm
        Identifier: 0x2e0
        Timestamp:  20535
        Subpixel:   unknown
        Clones:
        CRTC:       0
        CRTCs:      0
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter:
        non-desktop: 0
                supported: 0, 1
  1024x600 (0x2e1)  0.000MHz *current
        h: width  1024 start    0 end    0 total 1024 skew    0 clock   0.00KHz
        v: height  600 start    0 end    0 total  600           clock   0.00Hz
````

I'm assuming my "identifier" is "default". I have looked up many examples of the xrandr output and where mine says "default" another may say "HDMI-0" or similar.

I have set this up in my X11 config:

````
Section "Monitor"
    Identifier "default"
    # This identifier would be the same as the name of the connector printed by xrandr.
    # it can be "HDMI-0" "DisplayPort-0", "DSI-0", "DVI-0", "DPI-0" etc

    Option "Rotate" "inverted"
    # Valid rotation options are normal,inverted,left,right

    Option "PreferredMode" "1024x600"
    # May be necesary if you are not getting your prefered resolution.
EndSection
````

For reference, ````tvservice```` outputs:

````
state 0xa [HDMI DMT (87) RGB full 16:9], 1024x600 @ 60.00Hz, progressive
````

What am I doing wrong here?

My ````/boot/config.txt```` also ignores any lcd_rotate, or display_hdmi_rotate settings (which I know would not affect KlipperScreen).

alfrix, 2022-11-08:
Two very important things are missing:
1. Screen model (the manufacturer of the screen usually list the rotation to method to be used)
2. Raspberry model (rpi4 can't use display_x_rotate unless in legacy mode)

xrandr should say somthing like 
`DVI-0 connected primary 1600x900+0+0 (0x59) normal (normal left inverted right x axis y axis) 443mm x 249mm`
notice the rotation options listed at the right of the command
if it doesn't list rotation modes, you may need another method to set rotation

for example for official rpi display on DSI
https://www.raspberrypi.com/documentation/accessories/display.html#screen-orientation

> My /boot/config.txt also ignores any lcd_rotate, or display_hdmi_rotate settings (which I know would not affect KlipperScreen).

lcd_rotate and display_rotate are deprecated  and display_hdmi_rotate or display_lcd_rotate only work on pi3 or pi4 in legacy mode

see https://www.raspberrypi.com/documentation/computers/config_txt.html#display_hdmi_rotate

ajquick, 2022-11-09:
I have added the following line to ````/boot/cmdline.txt````:

````
video=HDMI-A-1:1024x600M@60,margin_left=0,margin_right=0,margin_top=0,margin_bottom=0,rotate=180
````

This is now working and I am also seeing ````xrandr```` also now correctly outputting this:

````
Screen 0: minimum 320 x 200, current 1024 x 600, maximum 7680 x 7680
HDMI-1 connected primary 1024x600+0+0 (0x42) normal (normal left inverted right x axis y axis) 0mm x 0mm
        Identifier: 0x40
        Timestamp:  20172
        Subpixel:   unknown
        Gamma:      1.0:1.0:1.0
        Brightness: 1.0
        Clones:
        CRTC:       0
        CRTCs:      0
        Transform:  1.000000 0.000000 0.000000
                    0.000000 1.000000 0.000000
                    0.000000 0.000000 1.000000
                   filter:
        Broadcast RGB: Automatic
                supported: Automatic, Full, Limited 16:235
        bottom margin: 0
                range: (0, 100)
        top margin: 0
                range: (0, 100)
        right margin: 0
                range: (0, 100)
        left margin: 0
                range: (0, 100)
        link-status: Good
                supported: Good, Bad
        CONNECTOR_ID: 89
                supported: 89
        non-desktop: 0
                range: (0, 1)
  FIXED_MODE (0x42) 49.000MHz +HSync +VSync *current +preferred
        h: width  1024 start 1064 end 1168 total 1312 skew    0 clock  37.35KHz
        v: height  600 start  603 end  608 total  624           clock  59.85Hz
````

Finally it is no longer referring to the display as "default" and is now HDMI-1 (though plugged into HDMI 0).

Next step to transform the touchscreen, but hopefully this helps someone with a RPi 4.

closed 2022-11-09 by ajquick