KlipperScreen/KlipperScreen #780: [BUG]package depandence error occurs in Install script

cliffmanw, 2022-11-08:
**Describe the bug**
package depandence error occurs in Install script when installing to mobian platform.
this is the package causing problem :"libfontconfig-dev" and " ibgirepository1.0-dev"

checked mobian-project.org, a related package "libglib2.0" do not exist.

But, as I know, months ago, the "KlipperScreen-install.sh" is runing OK at mobian.
the new "KlipperScreen-install.sh" I downloaded today is not working.

**Screenshots**
no screenshots

**Logfile**
When Runs to “PyGobject dependincies” stage, 404 error occurs, I modified file "KlipperScreen-install.sh",
removed package "libfontconfig-dev" and " ibgirepository1.0-dev" to make install.sh continue
But, installing still stops with error, log attached as follows

Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libfontconfig-dev is already the newest version (2.13.1-4.2).
The following additional packages will be installed:
  libglib2.0-dev libglib2.0-dev-bin
Suggested packages:
  libgirepository1.0-doc libglib2.0-doc libgdk-pixbuf2.0-bin | libgdk-pixbuf2.0-dev libxml2-utils
The following NEW packages will be installed:
  libgirepository1.0-dev libglib2.0-dev libglib2.0-dev-bin
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 1,787 kB/2,621 kB of archives.
After this operation, 22.6 MB of additional disk space will be used.
Do you want to continue? [Y/n]
Err:1 http://repo. unstable/main arm64 libglib2.0-dev-bin arm64 2.68.1-2mobian1
  404  Not Found [IP: 91.121.79.211 80]
Err:2 http://repo.mobian-project.org unstable/main arm64 libglib2.0-dev arm64 2.68.1-2mobian1
  404  Not Found [IP: 91.121.79.211 80]
E: Failed to fetch http://repo.mobian-project.org/pool/main/g/glib2.0/libglib2.0-dev-bin_2.68.1-2mobian1_arm64.deb  404  Not Found [IP: 91.121.79.211 80]
E: Failed to fetch http://repo.mobian-project.org/pool/main/g/glib2.0/libglib2.0-dev_2.68.1-2mobian1_arm64.deb  404  Not Found [IP: 91.121.79.211 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Installation of PyGobject dependincies failed (libgirepository1.0-dev gcc libcairo2-dev pkg-config python3-dev gir1.2-gtk-3.0)

#=======================================================#
 KlipperScreen installation failed!
#=======================================================#

---------------------------------------------------------------------------------------------------------------------------
after removed package "libfontconfig-dev" and " ibgirepository1.0-dev", and runs "KlipperScreen-install.sh"
showing follows

  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: pycairo
  Building wheel for pycairo (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/mobian/.KlipperScreen-env/bin/python /tmp/tmprihs1omw_in_process.py build_wheel /tmp/tmpzp8bbqql
       cwd: /tmp/pip-install-lm_bsq3d/pycairo_ceb19b28b86041d8b9f2709e86cf29a4
  Complete output (15 lines):
  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-aarch64-cpython-39
  creating build/lib.linux-aarch64-cpython-39/cairo
  copying cairo/__init__.py -> build/lib.linux-aarch64-cpython-39/cairo
  copying cairo/__init__.pyi -> build/lib.linux-aarch64-cpython-39/cairo
  copying cairo/py.typed -> build/lib.linux-aarch64-cpython-39/cairo
  running build_ext
  Package cairo was not found in the pkg-config search path.
  Perhaps you should add the directory containing `cairo.pc'
  to the PKG_CONFIG_PATH environment variable
  No package 'cairo' found
  Command '['pkg-config', '--print-errors', '--exists', 'cairo >= 1.15.10']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for pycairo
Failed to build pycairo
ERROR: Could not build wheels for pycairo which use PEP 517 and cannot be installed directly
Error: pip install exited with status code 0
Unable to install dependencies, aborting install

alfrix, 2022-11-08:
the dependencies are needed for PyGObject described on their site for debien/ubuntu
See https://pygobject.readthedocs.io/en/latest/getting_started.html#ubuntu-getting-started
This is not a KlipperScreen BUG, you need fix your distro probably change mirrors because those packages are necessary and need to be installed or KS can't work

guinevereperez, 2024-06-05:
> the dependencies are needed for PyGObject described on their site for debien/ubuntu See https://pygobject.readthedocs.io/en/latest/getting_started.html#ubuntu-getting-started This is not a [...]

Thank you for suggesting it I was able to fix thru homebrew

guinevereperez, 2024-06-05:
**This is what worked for me:**

1. Install homebrew for Linux: [](https://docs.brew.sh/Homebrew-on-Linux )

2. Brew install pygobject3 gtk4

4. Change the working directory to where your hello.py script can be found

5. Run python3 hello.py