KlipperScreen/KlipperScreen #785: [BUG] Screen not entering print status page

jan3ll3, 2022-11-09:
Something between 0.2.7-11 and 0.2.8-36 seems to have broken the ability to enter print status screen. I start a print and it does not come up, and have no way of even canceling the print from the screen afterwards. 

Was running klipper v0.10.0-289 but upgrading to klipper v0.10.0-623 fixed the issue.

alfrix, 2022-11-09:
a bug report like this it doesn't really help me, unless you can provide the KlipperScreen.log from when this happened (see troubleshooting)[https://klipperscreen.readthedocs.io/en/latest/Troubleshooting/] the new logs retain quite a bit of information, it may include a couple of days back

jan3ll3, 2022-11-09:
Apologies, I'm running this in a production environment and am quickly scrambling to get 3 machines that have recently been updated to klipperScreen working. 2 worked fine on the 0.2.7-1 & 0.2.7-11 version but i have 3 that will not enter the print status screen. I've since patched to the newest klipper install on one and they are working. I dont have a log handy, but it does look like there was a recent commit dealing with print status which I imagine breaks with the older version of klipper. Unfortunately that is all the info I have for you at this time

alfrix, 2022-11-10:
i suspect this bug (same as 750) was on the initial 0.2.8 ks batch, should be fixed, by now
closing reason: missing log

closed 2022-11-10 by alfrix