KlipperScreen/KlipperScreen #786: [FR] Proper SSL support and support for self-signed certificates

xorza, 2022-11-10:
**Is your feature request related to a problem? Please describe.**
I want my KlipperScreen to run on a system, separate from 3D printer. Imagine: printer in enclosure and KlipperScreen on a Raspberrypad (BTT Raspberry Pad 5) outside of enclosure. Both systems connected to the same wireless network. Moonraker set up to use port 7130 and self-signed cert.

Current implementation does not recognise 7130 port as ssl and tries to send HTTP requests. I edited python code to force https and got unverified certificate error, which I failed to overcome.

**Describe the solution you'd like**
Idea is to provide an additional flag which describes ssl behavior:
```
moonraker_host: 192.168.1.xx
moonraker_port: 7130
moonraker_ssl_mode: <disabled | selfsigned | secure>

```

**Describe alternatives you've considered**
Automatically try https if http failed and silently accept self-signed certs. - Sounds insecure.

**Additional context**
Basically I wanted to make my raspberrypad behave like a tablet and make it completely detached from printer.

alfrix, 2022-11-10:
Use port 443 and KS will switch to WSS

xorza, 2022-11-10:
@alfrix 
This is not a solution because:
1. It is never mentioned in docs that KS will treat 443 differently than 7125. Not evebody can read python.
2. default SSL port for moonraker is 7130, we should not confuse KS users tribal knowledge about port numbers
3. this doesnt solve self-signed cert issue

More thoughtful comments appreciated

closed 2022-11-18 by alfrix

xorza, 2022-11-19:
@alfrix thanks for the prompt changes. But does it support self-signed certificate though?