KlipperScreen/KlipperScreen #798: [BUG] Font size ignored

guestisp, 2022-11-20:
I'm almost sure that font size (in my case "large") is ignored everywhere except the setting (and even in setting i'm not 100% sure it works).

This is what I see at 800x480 on a 5'' LCD. Looks pretty small to me and there is no difference in setting "medium" or "small".
[image]

The issue is that with this font-size, is very hard to read something from 1 meter distance. (pic took at more or less 15cm)

(also, if you place "Elapsed" and "Remainin" on the same line, you free some space allowing to use a bigger font)

guestisp, 2022-11-20:
Is honored, but the default is still too small, to see an increase i had to set it to 1.20 but doing so results in bottom buttons out of screen.

I home someone increase the text and reduce the bottom buttons, they are too big even for a finger. (or some info could be removed in a second screen, like the extrusion speed and extrusion volume, I don't think would be useful to keep them always visible)

alfrix, 2022-11-20:
never open a bug request without a logfile, and this is a feature request as font size works you just want a point above the maximum

guestisp, 2022-11-20:
as wrote, i've found that font size works after opening the issue, when i've opened, i've tought It was a bug

commit 054523d9 (2022-11-21): settings: add extra large font close #798
closed 2022-11-21 by alfrix